OrcaSlicer/OrcaSlicer #6807: Repeatable CRASH on slice

brjohnso1, 2024-09-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 Beta

### Operating System (OS)

macOS

### OS Version

14.5

### Additional system information

Intel Mac

### Printer

Bambu P1S

### How to reproduce

Load the attached project.
Slice.

### Actual results

Crash

### Expected results

Should slice without crashing. 

### Project file & Debug log uploads

[SlushBusterPro58V3 v13.3mf.zip](https://github.com/user-attachments/files/17032371/SlushBusterPro58V3.v13.3mf.zip)
[debug_Tue_Sep_17_09_42_39_67121.log.0.zip](https://github.com/user-attachments/files/17032377/debug_Tue_Sep_17_09_42_39_67121.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2024-09-18:
Thank for reporting.
Does this file slice fine in previous versions?

Noisyfox, 2024-09-18:
The crash is caused by the empty first layer.

Noisyfox, 2024-09-18:
https://github.com/SoftFever/OrcaSlicer/commit/c36d4ad7c94b972b0bef2295c864f450008e91e5#diff-6ce31647f85fd26c50c87e8f851643fdff8a9fc3e38f3b65b4d2aec4b5bcf916

Seems to be caused by this change.

closed 2024-09-18 by SoftFever

brjohnso1, 2024-09-18:
Odd that it has an empty first layer. It prints on the Bambu slicer, and I used the "lay on face" tool to flip it just before slicing. Is the "lay on face" tool not placing objects all the way onto the bed?

brjohnso1, 2024-09-18:
I used “lay on face” before the slice/crash. Is “lay on face” not putting the object all the way down onto the printer bed?

SoftFever, 2024-09-18:
It does.

In this scenario, the bottom of the model is shaped like a 'V,' so the first layer ends up being a zero-width line on the bed. 
A few versions ago, Orca decided not to hide this issue with an 'auto-fix.' Instead, an error message is displayed to ensure users are aware of the problem with the model.

Since the 'auto-fix' function in BS or PS is still quite premature and unreliable, we believe it’s better to inform users about such issues directly.

PR #6817 (2024-09-18): Fix crash when first layer is empty
Fix #6807

All places that use `current_extruder_id` will first do a `if(current_extruder_id)` check, which will only be false if this optional is not set. But when it's initialized with `-1` all the checks will become useless which causes this crash.