OrcaSlicer/OrcaSlicer #6827: [toolchanger] Changing Global "Filament for Features" settings has no affect.

N3MI-DG, 2024-09-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Voron V2.4 Stealthchanger

### How to reproduce

1. Load a model
2. Under Multimaterial tab of Global Process change infill to another tool number.
3. Slice.

### Actual results

Modal gets sliced as a single tool print.

### Expected results

The infill would use a different tool than other features.

### Project file & Debug log uploads

[OrcaCube.zip](https://github.com/user-attachments/files/17055081/OrcaCube.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Adding a second model makes the prime tower appear, but after slicing the result is still single tool.
Changing the Filament for Features setting under Objects Process appears to work as expected.

mattiras, 2024-09-27:
Came here to look if reported, so I can confirm. Hope to get this fixed.

Within object level, function works as expected, if looking somebody islooking  for a temporary workaround.

mattiras, 2024-10-12:
Bug still present in 2.2.0-RC

Nick2121, 2024-12-24:
Came here to see if this was a known issue. Can confirm those settings have no effect. This should be a priority item now that faster multi-material printers are becoming available. This would actually be useful.

mattiras, 2025-03-25:
Problem is still present in 2.3.0.

closed 2025-10-01 by github-actions[bot]