OrcaSlicer/OrcaSlicer #6828: Print Order: By Color

Culain, 2024-09-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When there are multiple parts on the plate with different colors there are two issues:
- print by layer will change color on every layer leading to a lot of wasted material
- print by object has it so the printspace is very limited

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

In this case it would be nice to have a "hybrid" solution. Parts with the same material/color will be printed "by layer" and groups of parts with the same color are printed as "by object".

[image]

in this case, the two blue parts could be placed directly next to each other and printed as one part. after that they will be treated as "one object" on the plate for collision avoidance for the last (white) part. this way, this model could be printed twice or more on the same plate.

discip, 2024-09-19:
@Culain 
Just select both blue parts, right-click one of them and choose **`Assemble`**.

Prior to assembling the parts, you might want to place them close to each other.

But even after assembly, you can still move them individually by selecting the corresponding part from the list on the left in _`Object-mode`_.

FireflyRobots, 2024-10-26:
I see your point, but this is not the same as grouping by color without assembling them. Please attempt to set different support types or styles for assembled parts...

SPGWhistler, 2025-05-16:
The problem stated is probably not a great fit - but the idea here is a good one. Let me explain:

Print by Hybrid. A combination of by layer and by object.

The basic idea is:
Print by layer to a point, switch to next object, repeat.

Here is a detailed description:
Print each object one at a time (like by layer) until you get to the point where the nozzle (or other printer parts) would collide with other object. Then, switch to the next object and print its layers up to that same layer height. Loop for each object. Once all objects were printed to the same layer height, go back to first object and repeat.

For example:
I have two objects, A with 50 layers, and B with 30 layers. I want to print them at the same time, but I don't want to do layer by layer because of the travel moves and potential for stringing and errors in the print. (Filament cools quickly and could curl up, causing print failures, etc). Or, they are multi-material but have similar change heights - so I want to reduce the filament changes as much as possible.
The objects are placed on the bed spaced like they would be for by object mode.
The printer prints layers 1 to 20 of object A.
It then prints layers 1 to 20 of object B.
It then prints layers 21 to 30 of object B.
It then prints layers 21 to 50 of object A.

Obviously this might not work for all types of printers. But it could be a really useful feature because it would dramatically speed up printing time for multiple objects, while at the same time reducing the number travel moves between objects.

RafaelSpagnollo, 2026-01-02:
> The problem stated is probably not a great fit - but the idea here is a good one. Let me explain:
> 
> Print by Hybrid. A combination of by layer and by object.
> 
> Pros: You can print larger objects "by object" than your printer would normally allow. You can potentially reduce filament changes for multi-material prints. You can will have quite a bit less stringing and traveling than if you did it 'by layer'.
> 
> Cons: You might get 'lines' where the printer switches between objects. (Although I'm not sure on this.)
> 
> The basic idea is: Print by layer to a point, switch to next object, repeat.
> 
> Here is a detailed description: Print each object one at a time (like by layer) until you get to the point where the nozzle (or other printer parts) would collide with other object. Then, switch to the next object and print its layers up to that same layer height. Loop for each object. Once all objects were printed to the same layer height, go back to first object and repeat.
> 
> For example: I have two objects, A with 50 layers, and B with 30 layers. I want to print them at the same time, but I don't want to do layer by layer because of the travel moves and potential for stringing and errors in the print. (Filament cools quickly and could curl up, causing print failures, etc). Or, they are multi-material but have similar change heights - so I want to reduce the filament changes as much as possible. Or they are too tall and 'by object' printing would be impossible. The objects are placed on the bed spaced like they would be for by object mode. The printer prints layers 1 to 20 of object A. It then prints layers 1 to 20 of object B. It then prints layers 21 to 30 of object B. It then prints layers 21 to 50 of object A.
> 
> Obviously this might not work for all types of printers. But it could be a really useful feature because it would dramatically speed up printing time for multiple objects, while at the same time reducing the number travel moves between objects. And probably most importantly - it would allow you to print 'by object' tall objects which would normally be impossible.

I think my reply got exactly what you were trying to explain