OrcaSlicer/OrcaSlicer #6839: Printer stops heating before he was ready

jensus11, 2024-09-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta

### Operating System (OS)

macOS

### OS Version

Sequoia 15.0

### Printer

IDEX self build

### How to reproduce

First, I sliced my object and created a gcode file. The Printer printed the gcode very well, but at the half last layer he stopped the heating and printed the rest without heat.
He has in the gcode the line, but I don't know where he got it.
G10 SO ; preheat T-1 time: 30s

[image]

### Actual results

Bad last layer

### Expected results

  don't stop heating

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

HaythamB, 2024-11-07:
OK did some tracing, and I found this to be probably caused by the M104.1 handling for XL printers, specifically the "export_lines" code block should be moved into the IF statement preceding it to handle cases where the tool index is "-1".

In the below screenshot from the current release (V2.2.0), within GCodeProcessor.cpp in libslic3r>Gcode, lines 4920 to 4967 should be moved to start at line 4919 (ie within the IF code block right before it.

[image]

psychotik2k3, 2024-12-21:
is it merged yet ?

HaythamB, 2024-12-25:
Nope, no action taken yet. I have a windows compiled version with the fix applied, and I've confirmed its working. Guess its just a matter of time.

edspeds, 2025-01-25:
Any action on this yet?  Don't want to let it close without being addressed.

psychotik2k3, 2025-01-25:
> Nope, no action taken yet. I have a windows compiled version with the fix applied, and I've confirmed its working. Guess its just a matter of time.

can't you make a PR so it's included in the next release

edspeds, 2025-01-25:
I believe HaythamB did, I too have his fix and it's working well for me.

psychotik2k3, 2025-01-25:
nope code is still the same, still not in the if section above.
@HaythamB wana make the PR for you fix please ?

HaythamB, 2025-01-25:
https://github.com/SoftFever/OrcaSlicer/pull/7405

I did one a long time ago.. Do I need to update something to have it merged?

psychotik2k3, 2025-01-26:
ok read the Pr and comments it's in workflow.
thanks for the work.

commit d9c27f3d (2025-04-20): Fix #6839 with final tool preheating on multitool machines causing in… (#7405)

Fix #6839 with final tool preheating on multitool machines causing in appropriate temp settings

Seems like Orca is trying to preheat the next tool in a multitool print, and ends up calling a heater off command in the last 30 seconds of any print.

This happens because there's no handling to check if the next active tool is an actual valid tool index, or its a T-1 command to end the print  since we're using the last tool.

Simply moved the preheat commands into the conditional IF that automatically fixes this issue [... cut]
closed 2025-04-20 by Noisyfox