OrcaSlicer/OrcaSlicer #6840: Theatre/Widescreen Camera Mode

Grochocinski, 2024-09-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I like to have Orca Slicer pulled up on the side while I work so that I can view the camera feed in the device window and ensure the print completes successfully. However, since the camera window and control window are always side-by-side, the camera window can be quite small and hard to see, thus making it hard to monitor my prints without making the Orca Slicer window fullscreen.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I'd like the option to expand the camera feed on the device tab to make the camera feed easier to see.

### Describe alternatives you've considered

1. An option to stretch the camera feed across the top (similar to Theatre Mode on YouTube)
2. A button to pop out the camera feed into its own window (similar to the call popout option on Discord)
3. Having each section in the Device > Status window (Camera, Control, and Print Progress) be rearrangeable and resizable

GammelSami, 2024-10-02:
I think a better solution would be a "Pop-out-camera" button that creates an additional window with just the camera stream.
Pros:

- If the user wants it in full screen, they can maximize the second window.
- If the user wants it on their second monitor they can place it there how they like.
- ...while still being able to use the slicer software simultaneously.

drx6x, 2024-12-18:
yeah a pop out camera would be amazing so i can easily organize all the machine feeds I'm watching on my 3rd monitor.