OrcaSlicer/OrcaSlicer #6848: Main application window not appearing on macOS

Ah-Tui, 2024-09-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

macOS 15.0 (24A335)

### Additional system information

MacBook Pro 14 M1 Pro
Memory: 32GB

### Printer

Bambu Lab X1C

### How to reproduce

1. Click on the application icon from the macOS Launchpad

### Actual results

1. Icon appeared in Dock
2. Splash screen appeared
3. Update prompt appeared
4. Bambu Lab Network plugin download appeared
5. Only blank window appeared when using the "Show all window" in macOS
[image]

### Expected results

1. Icon appeared in Dock
2. Splash screen appeared
3. Main application screen appeared

### Project file & Debug log uploads

Unable to locate the OrcaSlicer folder in Application Support

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Tried to remove the whole application, then reinstall the application but same behavior exists. Also tried v2.2.0beta, ended with the same result.

TTRaven, 2024-09-29:
Going to add onto this, running into the same problem. The app opens, I can interact with the menu bar, open up settings (which does pop up), but the main window doesn't display

OrcaSlicer Version
2.2.0 beta 2 (I have attempted this with many versions, even using x86 via Rosetta)

Operating System (OS)
macOS

OS Version
macOS 15.0 (24A335)

Additional system information
MacBook Air M3
Memory: 8GB

But a work around I have found is to fullscreen that application, it freaked out for a second, threw back to 'Desktop 1'. I 3 finger swiped (or go into mission control and it will be at the top) and then Orca will display properly.
I was able to exit fullscreen and it proceeded like expected, and I have been able to relaunch Orca and it hasn't caused the original issue since.
Possible it is a weird result of MacOS fullscreen, and when the application is closed... Unsure at the moment.

Ah-Tui, 2024-10-03:
> Going to add onto this, running into the same problem. The app opens, I can interact with the menu bar, open up settings (which does pop up), but the main window doesn't display
>
> OrcaSlicer Version [...]

Thanks for the workaround! You saved my day!

Unfortunately, my problem still exist after relaunching Orca, but at least I can now use the Zoom (or any other window) button on the Menu Bar to get it open.

Hopefully this issue will get noticed and fix soon.

closed 2025-01-10 by github-actions[bot]

J0kerFace, 2025-04-11:
I have the same issue with 2.3.0 and 2.3.1 since macOS 15.4. 
fn + f didnt work for me as workaround.