OrcaSlicer/OrcaSlicer #6852: Brims not appearing in Preview Window

TwistedNexus, 2024-09-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-Beta

### Operating System (OS)

Windows

### OS Version

OS Name	Microsoft Windows 11 Home - Version	10.0.22631 Build 22631

### Additional system information

CPU: 13th Gen Intel(R) Core(TM) i7-13700H, 2400 Mhz, 14 Core(s), 20 Logical Processor
RAM: 32.0 GB
GPU: NVIDIA GeForce RTX 4080 Laptop

### Printer

Creality Ender 3 V3 KE

### How to reproduce

1. Open any model in OS
2. Select any brim option
3. Brim is not shown in the preview window but it does print

[image]

### Actual results

Brim is printed even though it is not shown in the preview window.

### Expected results

The brim should be represented in the preview window to see how large it is and if it will interfere with another part of the print or be over the edge of the print area.

### Project file & Debug log uploads

[Ender 3 V3 KE Side Spool Holder v1.zip](https://github.com/user-attachments/files/17082616/Ender.3.V3.KE.Side.Spool.Holder.v1.zip)
[log.zip](https://github.com/user-attachments/files/17082622/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Ergonomicmike, 2024-09-21:
According to your screenshot, you don't have Brim selected to display.

discip, 2024-09-22:
@TwistedNexus
> According to your screenshot, you don't have Brim selected to display.

That's exactly it!
Just set the tick to display Brim. 😊
[image]

TwistedNexus, 2024-09-23:
I toggled that a few times and must have taken the screenshot when I had it unchecked. The skirt placement was the exact same with the brim option set to off, auto, and outer only with the brim width set to 5, 8, 10, and 12mm.

I ended up reverting to 2.1.1, so I can't test it anymore, but I did account for that before submitting the bug report.

Thanks!

discip, 2024-09-23:
@TwistedNexus
> but I did account for that before submitting the bug report.

I'm not sure about that.

As this is the result when I open the project file you provided using the latest 2.2.0-beta:

[image]

Please close this issue, as it seems to be outdated. 😊

closed 2024-12-30 by github-actions[bot]