OrcaSlicer/OrcaSlicer #6858: Inconsistent line width

LAZER555, 2024-09-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 - beta

### Operating System (OS)

Windows

### OS Version

11

### Printer

elegoo neptune 2s with klipper and a lot of upgrades

### How to reproduce

I don't know why, but Orca gives me incosistent line width. It's the worst on a first layer. While Cura always has identical consistent lines. I have used Orca for a long time, but now i have to return to Cura, even though orca is so much more convenient and better, but i'm tired of this inaccuracies. Maybe it's related to an issue that somebody here brought up that it would be great to have flow control over walls and infills.
I have the same line width settings in both slicers.

### Actual results

In Orca
[image]

### Expected results

In Cura
[image]

### Project file & Debug log uploads

[medal1.zip](https://github.com/user-attachments/files/17084210/medal1.zip)
[log.zip](https://github.com/user-attachments/files/17084211/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

vgdh, 2024-09-21:
looks like the nozzle is too close to the bed. May be you have some z-offset configured in cura?

LAZER555, 2024-09-21:
> looks like the nozzle is too close to the bed. May be you have some z-offset configured in cura?

No. Just the same catrographer scan. The problem is not only the first layer. The lines are always inconsistent. It's like flow changes depending on the speed, but cura can compensate for it. I don't know

cubitik, 2024-09-24:
Are you using Arachne as wall generator? Have you tried using Classic as a wall generator?

LAZER555, 2024-09-24:
> Are you using Arachne as wall generator? Have you tried using Classic as a wall generator?

Yes, i use classic generator

closed 2025-01-01 by github-actions[bot]