OrcaSlicer/OrcaSlicer #6864: Pre-heat doesn't work on RepRapFirmware

DHM-projects, 2024-09-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 beta

### Operating System (OS)

Windows

### OS Version

Windows 10/11

### Additional system information

-

### Printer

Custom Multitool printer (ToolChanger with RepRapFirmware)

### How to reproduce

1 - Enable pre-heat based on time estimation in OrcaSlicer.
2 - Use the generated G-code on a ToolChanger running RepRapFirmware.

### Actual results

The pre-heat temperature is set as "active" temperature (G10 P{tool} S{active_temp}), meaning the tool on "standby" mode before the tool changing, doesn't pre-heat. It starts heating only when the tool is selected for use and so change the status from "standby" to "active". This means that the pre-heat time is not considered at all.

### Expected results

The tool should pre-heat before being selected for usage, while it is in the "standby" status. 

Based on our testing, this can be achieved using the following:
1 - G10 P{tool} R{standby_temp} S{active_temp}: need to set the same temperature for "R" and "S" values (pre-heat temperature).
However, it should be noted that this command is expected to be deprecated in the future and replaced by M568.

2 - M104 T{tool} S{target_temp}: sets both the active and standby temperatures to the target temperature. 
Where {target_temp} is the pre-heat temperature.
However, this command is deprecated in RRF for multi-tool machines and replaced with M568.

3 - The best solution: modifying the tool status from "standby" to "active" (need to change the status only, and don't run any tool change macro files) and set Tool temperature using M568 S{active_temp} A2. 
Where {active_temp} is the pre-heat temperature.

In the meantime, a GCode replacement tool in the Slicer can solve this issue.

### Project file & Debug log uploads

G10 S260 P1 ; preheat T1 time: 15s
need to be replaced with 
M568 S260 P1 A2 ; preheat T1 time: 15s
[Cubo_0.2mm_ABS_3h23m.zip](https://github.com/user-attachments/files/17085921/Cubo_0.2mm_ABS_3h23m.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Superbrain8, 2024-09-22:
would be also intresting to keep compability in the Future, RRF will eventually Deprecate G10 for setting temperatures.
M568 allows to change the state of the heater by just sending M568 P[Toolnumber] A1 for standby and A2 for active
for initially setting the standby and printing temperature the same command will be used with 2 more parameters added

issue with implementing will be quite likely the way temperature changes are currently handled
where the postprocessor just sets standby and active temperatures by adding G10 commands
switching it to M568 with respecting the A parameter would be a bit more involved

DHM-projects, 2024-10-08:
That's a great point regarding future compatibility.
We agree that the best solution would be to use the M568 command.

@SoftFever In the meantime, since implementing this might require a significant amount of work, perhaps we could use a GCode replacement tool in the slicer, similar to what PrusaSlicer offers? 
This feature could help address various small compatibility bugs while developing new features in OrcaSlicer.

closed 2025-01-15 by github-actions[bot]