OrcaSlicer/OrcaSlicer #6865: RegExp based G-Code Substitution like in PrusaSlicer

DHM-projects, 2024-09-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It's related to some extents to:
#6863 - Auxiliary Fan pin cannot be set 
#6864 - Pre-heat doesn't work on RepRapFirmware 
#6773 - Add Z coordinate when come back to purge tower

There are also many other small customizations that can be improved and bugs that can be worked-around by substituting G-Code after generation.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Porting the existing G-Code Substitution feature from PrusaSlicer to OrcaSlicer.

### Describe alternatives you've considered

We considered using the **Script Post-Processing** feature, but it's **not a portable solution**. 
It requires additional software to execute external scripts, which need to be put in a specific folder. 
This makes harder to share a complete printer profile.

### Additional context

We are aware of two related issues (#2235 and #903), but they were either closed after addressing only the primary issue or completely ignored. 
Therefore, we are opening a new issue. This feature would be highly useful in ensuring OrcaSlicer works as expected, even if there is a low-priority bug or missing feature. 
It would also be much easier than relying on **Script Post-Processing** and especially **helpful** if you need **to share a printer profile on multiple computers**.

mattiras, 2024-09-25:
As I mentioned in #2235 
Yes this would be very useful, now I had to take harder way and use Post processing script, which makes sharing profiles difficult. (setting up scripts to work is not easy for everyone)

galaxyman7, 2024-11-07:
This would be great to have so I can adjust extrusion ratio for individual feature types.

mattiras, 2025-02-06:
This should not be closed, feature would be very useful as I stated earlier.