OrcaSlicer/OrcaSlicer #6866: Connect to a printer over USB

Indyjunk, 2024-09-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I cannot connect to my printer via a USB cable.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

One of the main features Cura has over Orca slicer right now is support for printing over USB / Connecting to a printer over USB. Currently, I have my printer sitting on my desk connected to my PC. Having the ability to print over USB through Cura is simple and easy, as I don't have to download Octoprint or any other interfacing software to make any connection. Allowing users to do the same in Orca would take a key feature that Cura has over Orca away allowing users who have a 3d printer on their desk the option to simply connect with a readily available solution with no extra software or hardware solutions. 

### Describe alternatives you've considered

1. My current workaround is using Orca to generate my print path and then send the code to my printer Via Cura. This would be an unscary step if Orca supported USB printer connections.

m2ma, 2024-10-20:
I agree, we need something like this, it can bring many users from Cura.

flickerfly, 2025-01-29:
I'd appreciate it for my Neptune 3 Pro.

SpiderMurphLagos, 2025-02-19:
There are thousands of users with older machines that can only use USB.
Without this feature they will have to go back to Cura.

PavelS2180, 2025-03-02:
This feature is truly one of the most essential for those who are not just hobbyists but professionals involved in the development, assembly, and even creation of 3D printers. OrcaSlicer is already a powerful tool that supports network connectivity, but the lack of USB connection significantly limits its versatility.

Many users simply do not have the ability to modify or upgrade their printers. For them, the only available connection methods are USB or SD cards. However, in 2025, relying on SD cards as the primary method of transferring data is outdated and highly inconvenient. SD cards should be used solely for temporary storage of print files or configurations, not as a mandatory workaround for transferring files. Constantly swapping SD cards, dealing with data loss, and the inevitable wear and tear of flash storage—these are all unnecessary hassles that could be avoided if OrcaSlicer supported direct USB connectivity.

Adding USB support would make OrcaSlicer a truly complete tool, capable of replacing an entire set of fragmented solutions. It would free users from the need to rely on external software just for testing and basic printer management, making the entire workflow more seamless and efficient.

Jwn5k, 2025-08-24:
I too would like to see Print Via USB to be implemented, as much as I love using print over network and SD cards, some very specific scenarios have risen where I have the need to print over a USB cable due to lack of functionality or resources. It bugs me that it has been in CURA forever and Orca Slicer does not have it yet.

ElementW, 2025-11-09:
@Vad0268 While implementing USB serial communication can be easy, doing so *correctly* and in a cross-platform way is significantly more complicated than a TCP/IP connection, as the former has no unifying API, but the latter has BSD sockets itself hidden by the fact [`PrintHost`](https://github.com/SoftFever/OrcaSlicer/blob/4501bf6f5727a52eca2a9bc2467a1629aaf06eb0/src/slic3r/Utils/PrintHost.hpp) and its implementations use `libcurl` since it's all HTTP.

Beyond the variety of how the serial port can be accessed (`COM` ports on Windows, `tty`/`cu` devices on Linux/macOS, direct `libusb`-based access, maybe even HIDAPI in some cases), it:
* does not identify itself uniquely for the printer (easily mixed up with any other serial interface, even non USB; [look at the bullshit Cura has to do](https://github.com/Ultimaker/Cura/blob/34a542a7b58c4e3754e6be8971af1764835df4e1/plugins/USBPrinting/USBPrinterOutputDeviceManager.py#L140) or the disastrous, still very incomplete [`Serial`](https://github.com/SoftFever/OrcaSlicer/blob/4501bf6f5727a52eca2a9bc2467a1629aaf06eb0/src/slic3r/Utils/Serial.cpp) here in Orca that's used for Flashforge printers)
* has non-trivial configuration (baudrate really) for *any* data to flow through, and some virtual USB-serial adapters do not adhere to the USB CDC ACM spec and behave badly when their config gets changed
* may require elevated privileges to access, or an install step to lower the required privs (`udev` rules on Linux and whatnot)
* has no standard encoding, framing or error correction and associated retry algorithms
* some require the GCode to be trickled in ([see Cura](https://github.com/Ultimaker/Cura/blob/34a542a7b58c4e3754e6be8971af1764835df4e1/plugins/USBPrinting/USBPrinterOutputDevice.py#L383)) as they do not have enough buffer space for a full print, meaning baby-sitting [similar to what `TCPConsole` currently does](https://github.com/SoftFever/OrcaSlicer/blob/4501bf6f5727a52eca2a9bc2467a1629aaf06eb0/src/slic3r/Utils/TCPConsole.cpp) for Flashforge and MKS

Reminder that OrcaSlicer is free software and you are not entitled to any developers' time, the availability of the USB serial feature being only a function of somebody 1. having one such printer and 2. being motivated to implement it; it has nothing to do with your conspiratorial drivel.