OrcaSlicer/OrcaSlicer #6870: Tree supports ignore bottom interface and bottom z distance

Stormwing01, 2024-09-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Cpu: Ryzen 5600
Ram: 64Gb ddr4 3200mhz 
Gpu: RTX4060Ti 16gb

### Printer

Customized ender 3

### How to reproduce

Enable tree supports. Set style to "slim". Make sure that the model you're printing has tree supports resting on the model itself. Set top and bottom interface layer count to 3. setting bottom to "same as top" will suddenly eliminate all bottom interface, regardless of top layer count.
You will notice that the bottom of the tree supports that test on the model either has no interface and ignores bottom z distance, meaning it fuses with the model itself, or it will print the tree support, then suddenly print an incorrect amount of interface layers in the middle of the bottom tree support, then start printing tree supports again.

### Actual results

The tree support will be printed directly onto the model, fusing it to the model permanently.

### Expected results

Both the top and bottom of the support should have the correct amount of interface layers I have selected, as well as respect the Z distance I have set to make the supports not fuse to the model.

### Project file & Debug log uploads

The file has both the 3mf file and the log included.

[Bug report.zip](https://github.com/user-attachments/files/17089295/Bug.report.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]

Stormwing01, 2024-09-23:
Just finished a print and yeah, the supports really are just permanently fused to the model. here is a picture to show the bug's end result in an actual print.
As you can see, where I tried removing the support, it ripped out parts of the print, and in other parts, there is permanent scarring/outlining.

[image]

ThunderBoldt24, 2024-10-07:
I too am having this exact problem

tlhintoq, 2024-12-15:
2.3.0-dev - Latest nightly from 9 hours ago.
Still does it.  Totally ignores bottom interface to the model.
[image]

Pretty wasteful to have to do `[x] On build plate only` when its just a few CM of gap it could have done instead
[image]

tlhintoq, 2025-03-16:
Not stale.  Still does it 2.3.0-RC

closed 2025-06-23 by github-actions[bot]