OrcaSlicer/OrcaSlicer #6871: limitation for elephant foot compensation

Andreas-strg, 2024-09-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

if the elephant foot compensation is switched on and the part starts with an overhang, e.g. a chamfer or a radius, there is a great danger (the flow or the squish does not fit) that a path will be printed in the air

Typical error pattern:
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

a limitation of the elephant foot compensation in the area of the overhang in mm or i think better in degrees

vgdh, 2024-09-22:
It would be great,

Sineos, 2024-09-30:
Heh, I second this. Took me quite a while to figure out, why exactly this separation like in the image above was happening.

vgdh, 2024-12-30:
up

neo-jayfeather, 2026-01-03:
Based on the image, it seems more like a z-offset issue (?), as far as I can see. Elephant's foot compensation should align the first layer, meaning if you have issues, your elephant's foot compensation is off, or your z-offset is off (or both), as the goal of elephant's foot compensation is usually making the first layer the same size as the second layer. (It is more of a solution to a lower z-offset for better adhesion.) Please correct me if I'm wrong, but for cases like these consider reducing layer height, z-offset, or seeing if disabling the feature fixes it.