OrcaSlicer/OrcaSlicer #6872: Add option to lock "Auto rotate for arrangement" to 90° increments only

KittenRaee, 2024-09-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I mostly print functional parts using ABS and ASA and they quite rely on infill direction for their strength. I love the auto-arrange feature, but auto rotate option under arrangement options rotates pieces off the optimal direction rendering the option unusable for the use case.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add suboption under "Auto rotate for arrangement" that only autorotates parts by 90° (also implied by -90° and 180°).

### Describe alternatives you've considered

Painstakingly adjust arrangement for optimal fitment for many dozens parts manually at optimal orientations
Even more painstakingly than above let auto arrangement rotate pieces and adjust infill direction for every object.

ChaosBlades, 2024-09-30:
In addition to the above, it seems to favor 45 over 90 degrees even when it is completely unnecessary. 45 degree rotation should only be used if the part can not otherwise fit on the build plate for the reasons stated above.

xokahag, 2024-10-19:
In addition to the strength argument, print angle does also impact the surface finish of walls (VFA), and is therefore relevant for cosmetic parts too.

closed 2025-01-25 by github-actions[bot]