OrcaSlicer/OrcaSlicer #6876: False object "too tall" warnings in By Object print mode

otede, 2024-09-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

intel i3 32 GB RAM

### Printer

Bambu Lab A1

### How to reproduce

1. Load several object that should obviously fit in one print run.
2. Select By Object Printing mode

### Actual results

A warning sometimes pops up: Object <object_name> too tall, and collisions will be caused.

### Expected results

No such warnings as above, allow to slice, clear the gantry over any objects as necessary based on the printer profile gantry height given. The XY contour should be the only limiting factor for objects of height much less than the Z volume limits. 

### Project file & Debug log uploads

[Orca issue otede object too tall bogus warning.zip](https://github.com/user-attachments/files/17090822/Orca.issue.otede.obj
[20240922 log.zip](https://github.com/user-attachments/files/17090827/20240922.log.zip)
ect.too.tall.bogus.warning.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Related issue has been closed.
https://github.com/SoftFever/OrcaSlicer/issues/3299

Noisyfox, 2024-09-23:
You can always raise the "height to rod" value to your z limit and use it at your own risk. Slicer cannot gaurantee such clearance and plan a safety travel path for you, for example if your printer marco does anything stupid etc.

otede, 2024-09-23:
> You can always raise the "height to rod" value to your z limit and use it at your own risk. Slicer cannot gaurantee such clearance and plan a safety travel path for you, for example if your printer marco does anything stupid, or if you place your models parallal to the gantry etc.

I spent a solid year using Cura. No such issue was present. Any object could be printed in multi as long as both conditions were met:
1. Lack of XY contour collision
2. Objects' height less than Zmax - Gantry_height

Noisyfox, 2024-09-23:
> I spent a solid year using Cura. No such issue was present. Any object could be printed in multi as long as both conditions were met:
>
> 1. Lack of XY contour collision
> 2. Objects' height less than [...]

[image]
How would you ever print such arrangment in print-by-object mode using a core-xy printer without the gantry crashing into the first object?

otede, 2024-09-23:
> How would you ever print such arrangment in print-by-object mode using a core-xy printer without the gantry crashing into the first object (assuming you don't have a 20+cm long nozzle)?

Who said anything about core-xy? You can see above I stated BL A1 which is a bed slinger. And my old Cura powered scrap printer was also a bed slinger.

Noisyfox, 2024-09-23:
And:
[image]

Noisyfox, 2024-09-23:
> Who said anything about core-xy? You can see above I stated BL A1 which is a bed slinger. And my old Cura powered scrap printer was also a bed slinger.

Oh? Do you mean you could print that on an A1?

otede, 2024-09-23:
> Oh? Do you mean you could print that on an A1? You do realize it still have an X axis right?

If only you spent as much energy for evaluating the problem as you do playing a smarty pants. [pic22] Not only you went with an absolute extreme example, you also deliberately oriented the objects along the Y axis without the actual need to do so. In such a case it IS impossible to not have a collision. In my example - it is possible. [edit: It's simply out of the slicer feature set].

The Cura exmplanation popup is not clear. However I've just checked and I was indeed wrong with regard to Cura being satisfied with just meeting the "Objects' height less than Zmax - Gantry_height". It's just that my old printer had a gantry height some 2 times bigger than A1. Hence I routinely printed setups like this [example22].

I went ahead and filled Orca's plate with objects lower than my A1 gantry height of 25 mm and it let me slice.

I guess now we're in the "wishlist" area now as the currect behavior is "it is what it is." 

[pic22]
[image]

[example22]
[image]

closed 2024-09-23 by SoftFever

otede, 2024-10-16:
I've just discovered Prusa has already implemented some smarter algorithms for the purpose.
[image]

TheOverpassArsonist, 2026-01-01:
Sorry to potentially necro this but why was it marked as completed? Currently having a somewhat similar issue (I want to do a 'by objects' print with one short lid and one very tall container, which should easily by possible by printing the lid first THEN printing the container itself, but Orcaslicer is complaining that the container is too tall) and it seems like this was marked as 'complete' even though the comment immediately before that happened gave no indication of it being addressed.

[image]

Correct me if I'm wrong but a more 'complete' algorithm here would be sort of what's suggested here except checked incrementally by iterating through every object. E.g. : you can print an arbitrarily large number of short things, but only one tall thing. (because once the tall thing is printed, that interferes with future gantry movements) This has an exception if you carefully order your objects such that in printing each new one the gantry would never have areason to push over the other ones. (though on a raising bed design instead of a flying gantry design there is still an upper-height limit before the past prints physically hit the roof of the printer - if it's enclosed at least)

This actually raises an interesting alternative print mode, 'chunk by chunk' printing, where it goes object by object in multi-layer segments, then wraps back around. So it prints 20 layers of object 1, 20 layers of object 2, 20 layers of object 3, 20 more layers of object 1, 20 more layers of object 2, 20 more layers of object 30, etc. This would be a middleground between both layer by layer and object by object that still minimizes stringing, unnecessary travels, etc. without as many geometric considerations necessary. (for instance there would be no additional height limit, since at a macro-scale it's still going layer by layer, it's just working in multi-layer chunks)

otede, 2026-01-01:
@Azio-Pantheon Re chunks - you can already achieve this by using Assembly in right click menu.