OrcaSlicer/OrcaSlicer #6878: [BUG ]CUT-Gizmo: wrong flap-tolerance-gap-width

discip, 2024-09-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

latest 2.2.0-beta

### Operating System (OS)

Linux, Windows

### OS Version

not relevant

### Printer

not relevant

### How to reproduce

[image]

### Actual results

The gap width of the slanted part of the dovetail doesn't correspond to the set value.

### Expected results

Obviously the opposite of the actual result. 😊

### Project file & Debug log uploads

https://github.com/SoftFever/OrcaSlicer/blob/38cd758e948e5c2f9edddd5b281570b9e9cc0a8b/src/slic3r/GUI/Gizmos/GLGizmoCut.cpp#L728

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2024-09-23:
Does this happen on 2.1.1 as well?

discip, 2024-09-23:
> Does this happen on 2.1.1 as well?

Yes, unfortunately exactly the same behavior.

discip, 2024-09-23:
> Does this happen on 2.1.1 as well?

Yes, unfortunately exactly the same behavior.

**`BambuStudio`** and **`PrusaSlicer`** also have this issue.

Noisyfox, 2024-09-23:
I think the math works correctly here. The width tolerance here is not how much the two side faces are apart from each other perpendicular:
[image]

but rather how far apart horizontally:
[image]

And also, since the dovetail has two sides facing side ways, this tolerance is the amount with both sides addes up.

discip, 2024-09-23:
> And also, since the dovetail has two sides facing side ways, this tolerance is the amount with both sides adds up.

Yeah, that makes absolutely sense if you consider that between the other 2 horizontal faces there is no gap at all.

But that (Gap only on one set of opposing faces.) makes this approach of creating a groove somewhat fishy.

The better solution would be a gap that has a homogeneous distance between all opposing faces like the following, except without the radii (These are only to make clear that the distance between the slanted faces should be measured perpendicular to the faces).

[image]

I hope I was able to make myself clear.

thanks in advance

Noisyfox, 2024-09-23:
However, consider how FDM tend to be more inaccurate on XY direction than Z, it's somewhat make sense to calculate the tolerance on the xy direction.

discip, 2024-09-23:
Ah, that's a good point.

I was having a case in mind where the cut would be placed on the XZ or YZ plane so that the profile would lie on the XY plane and in this direction not Z direction would be involved.

Do you think it would make sense to differentiate between those cases?
I'm aware of the fact, that there would most likely be scenarios where someone would place a cut diagonally instead of vertically or horizontally. 😊

Would it be worth the effort?

Noisyfox, 2024-09-23:
Ah how stupid I am! I forgot that the cut plane can be tilted... Then my theory about the xy inaccuracy no longer holds. I think it might be better to do what you suggested, though this could be a breaking change and some might find their cut grooves no longer fit.

discip, 2024-10-04:
Is there a possible solution for at least the two cases I mentioned?

discip, 2024-10-04:
Is there a possible solution for at least the two cases I mentioned? 😅

closed 2025-01-10 by github-actions[bot]