OrcaSlicer/OrcaSlicer #6880: Automatically Flow Correct Infill/Wall Overlap

CCS86, 2024-09-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It is pretty standard to use the infill/wall overlap feature to minimize small gaps in this area. But, the problem with this is that this overlap is not flow compensated, so it leads to over-extrusion.

Because only the paths adjacent to the wall contribute to this over-extrusion, the magnitude depends on how close the walls are to each other. I believe that this is a huge part of what drove the development of "small area flow compensation". You can actually use this feature to flow compensate these overlapped paths:

[image]

But, as the angle between the infill and wall becomes more shallow, those overlapped paths become longer, and it's not practical to use small area flow comp to fix this constant amount of over-extrusion:

[image]

I'm proposing that infill paths which have the infill/wall overlap applied to them should be automatically flow compensated for the amount of overlap defined in the user setting. I think that this will detangle small area flow compensation and allow it to be fine tuned more accurately. It will likely allow you to use larger infill/wall overlaps and make it easier to remove all voids.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

...

igiannakas, 2024-09-23:
The monotonic lines pattern helps a lot with this.

CCS86, 2024-09-28:
> The monotonic lines pattern helps a lot with this.

It can, but there are a couple limitations / drawbacks:

- It can cause a ton of infill retraction at low angles
- You can't use the monotonic line pattern for sparse infill, so you have to cheat it in with top/bottom layers or solid threshold, which gives you less control over the print.