OrcaSlicer/OrcaSlicer #6885: Belt printer support on OrcaSlicer

vikichu, 2024-09-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No, this is an additional feature.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

There is an existing issue: Belt support #2628, but this issues isn't labeled as "Feature Request".
 
Slicer should have functionality to support belt printer of varying degrees. Typical is 45 degrees. GUI should render out sliced objects properly.

### Describe alternatives you've considered

1. Ideamaker as slicer: Useable, but not as good of an experience as OrcaSlicer/ PrusaSlicer derivatives.
2. Cura: Haven't tried this slicer for belt printer.

enzoav, 2024-10-29:
+1

As well as all of the +1's in this thread: https://github.com/SoftFever/OrcaSlicer/issues/2628

🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾🙏🏾

emanuele1412, 2024-11-22:
+1 here
Belt printers are getting more popular and with the klipper enabled IR3 V2 coming to the market, there will be a big demand for a good slicer that is capable to print at 45 degree angle.
Currently only Ideamaker slicer can do this but is extremely basic, slow, overcomplicated and featureless. An advanced slicer like Orca should could be lightyears ahead if it gets develepoment towards belt printing and would revolutionize the marlet of such software.

Jerichoulr, 2024-11-23:
+1 what do we have to do to get this implemented is there a way to "donate" to get this feature added? I would love to use orca slicer with the idea former ir 3 v2

enzoav, 2024-11-30:
> +1 what do we have to do to get this implemented is there a way to "donate" to get this feature added? I would love to use orca slicer with the idea former ir 3 v2

I too will donate whatever is needed to get this ball rolling.  Belt printing needs Orca Slicer to save it. 🙏🏾🙏🏾🙏🏾

D3printingFL, 2024-12-19:
What do we have to do to get this more attention?

enzoav, 2024-12-21:
🤷🏾‍♂️

xboxhacker, 2024-12-27:
+1 
+1 
+1 
I have 3 belt printers.

enzoav, 2025-01-05:
> What do we have to do to get this more attention?

start throwing money at someone...

Jerichoulr, 2025-01-06:
Who do we have to throw money at?

basz, 2025-01-09:
I think it's not *that* complicated. before slicing a matrix transformation needs to be applied (taking the angle into consideration). I have played with this to figure out how it could work in threejs.

Then, when viewing the sliced gcode the viewer should apply the reverse of the original matrix before rendering so that the geometry is back into its original form.

First layers modifications should be applied in a post processing of the gcode step of the gcode. (as on a belt printer the first layer is actually accross many layers.

I'm sure I'm missing some things but that should be the theory...

Justjohn1982, 2025-02-02:
+1 for this feature!

bathtub-toast, 2025-02-06:
+1 from me as well, somewhat motivated by the IdeaFormer IR3 v2 running a test print behind me. Also motivated by how much better an experience Orca has been for me compared to Cura, I refuse to go back.

martinjohuber, 2025-02-09:
+1 would love to see beltprinter support, if you need any support I can help you

grzbiet, 2025-02-11:
+1 and a question - what do you recommend for slicing while we wait for orca infinite?

Alaric323, 2025-02-11:
> +1 and a question - what do you recommend for slicing while we wait for orca infinite?

For now, the only major slicer that has belt support built in appears to be IdeaMaker.  Most other implementations are bodged together versions of Cura, like Blackbelt Cura and Creality Slicer.

actinaria, 2025-02-14:
Definitely would love to bee belt printers integrated into Orca

KazeOne, 2025-02-20:
I too would like to see such support, would even donate towards it's development

d-portero, 2025-02-23:
Someone  posted a GoFundMe to try and pool money to get this worked on https://www.gofundme.com/f/get-3d-belt-printers-into-orca-slicer

https://github.com/SoftFever/OrcaSlicer/issues/2628

xboxhacker, 2025-03-13:
If anyone wants to give it a try.....

I have revived an 8 year old script to allow you to slice on Orca and print on a belt printer. It can be found on my repo.

[link](https://github.com/xboxhacker/Tilted-Bed-Conveyor)

Download the release or build your own from the C# code.

This is currently working for me on my IR3V2. It _should_ work for the V1 and CR30 as well.

Leorizer, 2025-04-07:
+1 I have a IdeaFormer IR3 V2 and I realy ´dislike the experience of the IdeaFormer Slicer (It Succs) the only OK way of slicing for Belt printers i found is a Cura Belt plugin but it is on a older Cura version and Supports/Brim and other stuff doesn't work at all.
Id love to use Orca or Prusaslicer for the IR3 V2

tomfc262, 2025-05-20:
YES Please I use orca for 7 bambu machines and just bought a ideaformer-3d ir3 v2 for repeating prints want to keep everything under one slicer that just plain works

nikitax75, 2025-05-20:
+1 Have a printfarm with 20 printers and among them 4 IR3V2, probably one the best core XY we had, with the belt it's a killer printer. Just works well, fast and is reliable. Orca needs to support this printer, they're selling them by the hundred

hegge25, 2025-06-13:
+1. Yes please, would love to use orca for my ideaformer IR3V2

csscsccs, 2025-08-01:
> If anyone wants to give it a try.....
> 
> I have revived an 8 year old script to allow you to slice on Orca and print on a belt printer. It can be found on my repo.
> 
> https://github.com/xboxhacker/Tilted-Bed-Conveyor
> 
> Download the release or build your own from the C# code.
> 
> This is currently working for me on my IR3V2. It _should_ work for the V1 and CR30 as well.
> 
> **This is just a band-aid!!** _You can not use slicer supports with this._ You can model them into your STL tho.
> 
> ## We still need 100% native Orca slicing.

when i use it's gcode,I always got this error"!! Move out of range: 97.203 -5.784 0.283 [75.697]"

neusilon, 2025-08-06:
Would using conical slicing, as seen in this CNC Kitchen video, be a good substitute until belt support is implemented?
https://www.youtube.com/watch?v=1i-1TEdByZY

petros-tinkers, 2025-09-18:
I would love to get belt support for orca. Orca is the best slicer ever.

Dear developers, If you decide to make this happen, make the angle configurable including negative degrees. I am building an experimental printer and tried using Ideamaker for testing but it wouldn't support my printers geometry due to this (it only accepts positive angle values).

marcusg101, 2026-05-07:
> +1 CR-30 User. Current OrcaSlicer user.

Yeah I'm using a custom ender 3 that I modded with a kit. It needs proper
software.

closed 2026-10-09 by SoftFever