OrcaSlicer/OrcaSlicer #6886: Black screen on Arch Linux

Pegoku, 2024-09-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2-Beta

### Operating System (OS)

Linux

### OS Version

Arch Linux

### Printer

None

### How to reproduce

I installed orca-slicer-unstable-bin (also tried with orca-slicer-bin, and Appimage) from AUR, and I get no info on the app.
[image]

### Actual results

I'm unable to use the app

### Expected results

Be able to use Orca

### Project file & Debug log uploads

```
[2024-09-23 22:33:30.532151] [0x00007fc4582bfc40] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:30.691: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:30.692: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:30.692: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:30.693: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:30.694: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:30.695: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:30.695: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:30.696: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:30.697: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:30.698: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:30.874: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:30.876: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:30.876: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:30.877: gtk_widget_set_size_request: assertion 'width >= -1' failed

[... 203 lines cut ...]

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:33.194: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:33.194: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:33.196: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:33.196: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:33.197: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:33.199: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:65727): Gtk-CRITICAL **: 22:33:33.199: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (orca-slicer:65727): CRITICAL **: 22:33:33.220: Cannot register URI scheme wxfs more than once

** (orca-slicer:65727): CRITICAL **: 22:33:33.220: Cannot register URI scheme memory more than once
Failed to create GBM buffer of size 820x660: Invalid argument
```

### Checklist of files to include

- [X] Log file
- [ ] Project file

silvergasp, 2024-09-24:
I also see something very similar on Nixos, with a similar error log. The major difference is that I can see some of the menu's but can't see the 3d window.

[image]

silvergasp, 2024-09-24:
Interestingly I was able to work around the issue by switching from gnome (on wayland) -> gnome (on X). Obviously that's not ideal and it should work fine on wayland too as Xserver is moving towards deprecation but it does help in the mean time.

CitrusIntellect, 2024-09-25:
Hey fellow Arch & hyprland user! I had this exact same problem but I figured out it was because of an environment issue, my solution was to modify my orca-slicer.desktop file to also run with the Environment Variables: __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json

I don't recommend running this in the hyprland config under 
env = __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json 

as that will modify all of the EGL Vendor library's for every app but modifying the .desktop file works great.

Hope this helps!

Pegoku, 2024-09-28:
> Hey fellow Arch & hyprland user! I had this exact same problem but I figured out it was because of an environment issue, my solution was to modify my orca-slicer.desktop file to also run with the [...]

That worked!!!
Thanks

CitrusIntellect, 2024-09-30:
Hey @Prgoku, glad it worked, I suggest you close the issue as it is just something stupid with nvidia lol.

closed 2024-09-30 by Pegoku