OrcaSlicer/OrcaSlicer #6887: Orcaslicer Ubuntu2404 Beta2 AppImage will not start

ckvsoft, 2024-09-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2 beta2

### Operating System (OS)

Linux

### OS Version

Ubuntu 22,04 / Debian 13

### Printer

VZBot

### How to reproduce

Start the AppImage

### Actual results

Speicherzugriffsfehler (Speicherabzug geschrieben)

### Expected results

should run

### Project file & Debug log uploads

[logfile_newinstall.zip](https://github.com/user-attachments/files/17104007/logfile_newinstall.zip)
[logfile_currentinstall.zip](https://github.com/user-attachments/files/17104015/logfile_currentinstall.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

[2024-09-23 22:22:54.668908] [0x00007f46fef714c0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
loading network module at /home/chris/.config/OrcaSlicer/plugins/libbambu_networking.so
after dlopen, network_module is 0x558578e9a020

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:55.604: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:55.605: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:55.607: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:55.607: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:55.610: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:55.611: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:55.612: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:55.613: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:55.614: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

[... 228 lines cut ...]
(orca-slicer:21135): Gtk-CRITICAL **: 22:22:57.233: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:57.233: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:57.251: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:57.252: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:57.270: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:59.250: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:21135): Gtk-CRITICAL **: 22:22:59.250: gtk_widget_set_size_request: assertion 'width >= -1' failed
1. lo    : 127.0.0.1
2. wlo1  : 10.1.1.63

Speicherzugriffsfehler (Speicherabzug geschrieben)

This is for my current installation

a cleanup of .config/OrcaSlicer -- newinstall ,than i can start Orcaslicer
io also can connect to my printer and after few seconds the same 
Speicherzugriffsfehler (Speicherabzug geschrieben)

All other Appimages i have will run sucessfull also with network access

ckvsoft, 2024-09-24:
I found out that it is the network plugin. When I connect one of my printers, OrcaSlicer crashes after a while. Even without doing anything. All you have to do is enter the IP address and run the test. After some time there is a memory violation. After that you can no longer start Qorcaslicer, or it crashes again shortly afterwards.

jeikashu, 2024-10-03:
I think you are correct. I did a little dig, and if you go to the folder config and find `name_of_your_printer_in_Orca_slicer.json` there is a `printer_host` line. I deleted it and now OrcaSlicer works as it should (I don't dare trying to connect the printer again lol)

klack, 2024-10-30:
Couldn't find the line, but delete my config folder worked perfectly!

deployor, 2024-12-25:
When will this be fixed? I do want to connect to my printer...

ckvsoft, 2024-12-26:
> When will this be fixed? I do want to connect to my printer...

Isn't that already resolved? For me, the Linux version
OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage
is now working without any issues on both Ubuntu 24.04 and Debian 13.

However, I don't understand why there are two different AppImages. The required libraries should all be included within the AppImage itself and not rely on the system it runs on.

deployor, 2024-12-28:
Still doesnt work to me, i can use Moonraker but when i set fluidd on my port and everything in my network it crashes...

deployor, 2024-12-28:
Its fine but it kinda sucks just seeing the moonraker welcome site on my device tab kinda sucks, could also be fixed by just including a little thing where u can set which site to use for the device tab!

michas79de, 2025-03-26:
Similar issue.
I can use Orca Slicer without a printer connection via USB stick.
I also can use OS, when I connect it with Octo4a, an android octoprint server.
But when I connect to my new printer with Rinkhals/Moonraker/Mainsail/Fluidd, OS immediately crashes and keeps crashing, when I try to start it, until I delete the config folder.
But entering the moonraker port 7125 helps.

ampp33, 2025-06-30:
I still see this issue, please don’t close it

closed 2025-11-04 by github-actions[bot]