OrcaSlicer/OrcaSlicer #6898: Spiral concentric infill

m-sundman, 2024-09-24:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The concentric often looks nice on the top or bottom, except for the jump the extruder makes when moving from one concentric ring to the next.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Instead of having multiple rings inside each other, as in the normal concentric infill pattern, it would make a spiral of the rings, so the extruder could print the whole area in a single line (or a few spirals, for more complicated areas).

### Describe alternatives you've considered

Archimedean Chords prints in a single line, but only for perfectly round areas.

m-sundman, 2025-02-28:
> Orca bot: this issue is stale because it has been open for 90 days with no activity.

Well, this same issue is talked about from related issues. Fixing this would fix them, too.

m-sundman, 2025-03-27:
#6510 would also be fixed if spiral concentric infill is implemented.

skl111, 2025-04-04:
Currently you can use bambustudio or and older version.

I would love to have and option to select the old concentric pattern so I don't have to use bambustudio.

m-sundman, 2025-07-16:
The problems caused by the lack of this feature is coming up quite often on discord and facebook and, well, basically wherever people talk about this slicer.

closed 2026-09-06 by RF47