OrcaSlicer/OrcaSlicer #6900: 3d preview blank when on wayland when using latest nvidia drivers

haileyy123, 2024-09-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Arch Linux

### Additional system information

GPU: Nvidia RTX 4070
CPU: Ryzen 5 5600
Memory: 48gb

### Printer

Bambu Lab A1

### How to reproduce

Go to the preview tab and the 3d preview wont show up
[image]

### Actual results

The 3d preview dosent show up

### Expected results

The 3d preview should appear

### Project file & Debug log uploads

[debug_Wed_Sep_18_05_34_53_5328.log](https://github.com/user-attachments/files/17125586/debug_Wed_Sep_18_05_34_53_5328.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

[image]

TMBL-DEV, 2024-10-25:
same issue

mabl, 2024-11-02:
Same for me on Nixos unstable.

justdimaa, 2024-11-02:
same on nixos unstable, nvidia 565.57.01

conorlmcbride, 2024-11-10:
same issue. Nvidia 565.57.01 on bazzite (fedora atomic).

OMNIDROID2995, 2024-11-14:
> same issue. Nvidia 565.57.01 on bazzite (fedora atomic).
> 
> edit: I found a temporary fix in a couple other related threads here. Uses software renderer, so performance is worse. set env variable `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json`

Ohhhh thank you very, very much. Is there no way to render the whole thing hardware-accelerated? Software rendering is actually sufficient for most print projects. I'll get to work on that. Greetings

Fr0sty4, 2024-11-16:
For anyone using the flatpak -- this env var should work for software rendering and stays inside the flatpak:

__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json

yodatak, 2025-02-15:
Its not working today to so don't stale issue

closed 2025-05-25 by github-actions[bot]

dalailahner, 2025-05-25:
sadge

TheOverpassArsonist, 2026-03-08:
can confirm, the new flatpak release works fine for me, full acceleration and everything.