OrcaSlicer/OrcaSlicer #6901: Step on the outer perimeter.

6Yuri6, 2024-09-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.2.0 and V2.1.1

### Operating System (OS)

Windows

### OS Version

11

### Additional system information

CPU: AMD Ryzen 5 7500F
GPU: AMD Radeon RX 6700 XT
Memory: 32 Gib

### Printer

Anycubic Kobra 2 Plus

### How to reproduce

Printed flowerpot with a shell thickness of 8 mm constantly get a step on the outer wall of the model. On overhanging surfaces it is expressed more than on straight ones. There is an improvement if I change the order of printing inner/outer but I can not completely overcome it. In OrcaSlicer itself the problem is not visible.
[image]
[image]
[image]
[image]
[image]
[image]

### Actual results

There is a step on each printed copy.

### Expected results

Smooth outer perimeters.

### Project file & Debug log uploads

[shell.zip](https://github.com/user-attachments/files/17126861/shell.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2024-09-25:
This is due to material shrinkage - its the infamous "benchy hull line" problem.

vgdh, 2024-09-25:
> This is due to material shrinkage - its the infamous "benchy hull line" problem.

I wonder can material shrinkage be calculated and compensated somehow.

igiannakas, 2024-09-25:
> > This is due to material shrinkage - its the infamous "benchy hull line" problem.
> 
> I wonder can material shrinkage be calculated and compensated somehow.

On a whole model basis, yes. On a feature basis, no...

Whole model compensation is in the filament settings (shrinkage XY and Z):
[image]
This helps if the model ends up being smaller than the intended dimensions because of shrinkage.

closed 2025-01-03 by github-actions[bot]