OrcaSlicer/OrcaSlicer #6907: Out of memory when splitting into parts

0kenx, 2024-09-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Ubuntu 22.04

### Printer

Bambu A1 mini

### How to reproduce

1. Click the assembly on the plate
2. Right click -> Split into objects
3. std::bad_alloc

### Actual results

"OrcaSlicer will terminate because of running out of memory."

### Expected results

normal operation

### Project file & Debug log uploads

[cableclips.zip](https://github.com/user-attachments/files/17139619/cableclips.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

closed 2025-01-01 by github-actions[bot]

konradmb, 2025-03-10:
The issue still persists in 2.3.0-beta2. Most often with multi-part .step files.

0h41, 2025-03-22:
Still an issue in 2.3.0 stable.