OrcaSlicer/OrcaSlicer #6908: Orca Slicer does not Launch on Arch

CitrusIntellect, 2024-09-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1-1

### Operating System (OS)

Linux

### OS Version

Arch linux

### Additional system information

Nvidia 4060

### Printer

None

### How to reproduce

install orca-slicer-bin from aur (using paru & yay), launch orca-sclier

### Actual results

[2024-09-25 16:53:49.725148] [0x000074adee955c40] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
[1]    202507 segmentation fault (core dumped)  orca-slicer

### Expected results

Orca Slicer launches and goes through setup.

### Project file & Debug log uploads

Launched Here

[2024-09-25 16:53:49.725148] [0x000074adee955c40] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
[1]    202507 segmentation fault (core dumped)  orca-slicer

Ended Here

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

It was working just fine before updating orca-slicer, tried to reinstall and it did not work.

CitrusIntellect, 2024-09-27:
It found out it was patched in Beta 2.2

esistgut, 2024-10-13:
```
$ Downloads/OrcaSlicer_Linux_V2.2.0-rc.AppImage 
[2024-10-14 00:07:07.608166] [0x00007be3405fdc40] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
Segmentation fault (core dumped)
```

The issue seems to be still affecting 2.2.0-rc.

CitrusIntellect, 2024-10-22:
The issue does still exist, #7099 Nvidia cards are the issue here

CitrusIntellect, 2024-10-22:
I can no longer get Orca slicer to crash, I used the 2.2.0 rc with this command:

__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json

And it just wont crash

trebory6, 2024-10-26:
> I can no longer get Orca slicer to crash, I used the 2.2.0 rc with this command:
> 
> __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json
> 
> And it just wont crash

So I tried this solution and it opens, however it still crashes when viewing the "Home" tab and the "Device" tab.

Also, while this issue might be related to having Nvidia cards, Nvidia cards are not the issue.

The fact that there are solutions using Environmental Variables means this is a software issue. These are bugs. The cards aren't the issue, the bugs that cause the incompatibility with one of the most widely used graphics cards brands are the issue.

CitrusIntellect, 2024-10-30:
I reinstalled arch because another OS messed up everything, it will crash

kohbo, 2024-10-30:
> The issue does still exist, #7099 Nvidia cards are the issue here

I have this issue and I'm using an AMD 7900xtx

kohbo, 2024-10-30:
Switching to the latest version on Github (2.2.0 beta 2) is working for me

CitrusIntellect, 2024-10-31:
I just compiled OrcaSlicer from nothing and it is working great! I would suggest that first.

closed 2024-10-31 by CitrusIntellect

padcom, 2024-11-04:
It fixed Orca for me too. I was able to not only upload but also the device tab shows up again and actually works. Thank you!

quartzar, 2025-08-16:
It is still woefully broken on Arch.

Anotic, 2025-08-19:
still completely broken for me too damn

bora89, 2026-01-28:
I have installed `orca-slicer-bin` and it finally started