OrcaSlicer/OrcaSlicer #6911: PA Pattern for adaptive mesh

Dave811, 2024-09-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When I calibrate the PA, KAMP does the Adaptive Mesh and Adaptive Purge.
But it then calibrates the whole bed and draws the purge line at the end of the bed.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Could it be, that the position that the excluded objects use, be calculated for PA Pattern too?

SphaeroX, 2024-09-27:
I can confirm this behavior, the whole bed was always leveled. I only use adaptive mesh, no KAMP.

staal54a, 2024-10-05:
I also see this issue and so do at least a few others per #3994 (never resolved).

Elias23, 2024-10-24:
Also see this issue, creates an annoying time sink on big printers.

Dave811, 2025-01-23:
This issue still exists, so it's not stale

Elias23, 2025-01-23:
Not stale, please consider fixing it.

Benik3, 2025-03-16:
It happens also with PA line test