OrcaSlicer/OrcaSlicer #6916: Undo on large projects *slow*

rkarlsba, 2024-09-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta

### Operating System (OS)

macOS

### OS Version

macOS Sonoma 14.7

### Additional system information

Apple M2 Pro
16GB RAM

### Printer

Bambu X1C Combo

### How to reproduce

Make a project. Copy/paste a ton of objects. Save the project (just in case) and edit a single object somehow and then undo. 

### Actual results

With my current project (in which I don't really know the total number of objects because I haven't manually counted and because of [#2715](https://github.com/SoftFever/OrcaSlicer/discussions/2715), a simple undo after rotating a single object, takes close to 30 seconds.

### Expected results

Undo really should be _**way**_ faster than this. It's a modern laptop, after all.

### Project file & Debug log uploads

[daabloe-bambu.zip](https://github.com/user-attachments/files/17148664/daabloe-bambu.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

No logfile added, since orcaslicer doesn't crash

rkarlsba, 2024-10-03:
Also, if working on large projects (like the one attached) and adding parts to a plate, a shift+A to arrange the parts locally on that plate, can take half a minute without problems. With a smaller project, it's done in milliseconds. I can only guess that this is related.

rkarlsba, 2025-01-02:
It might be stale, but the problem is still very much the same. Please fix this!

IanVS, 2025-01-27:
I wonder if this is only a problem on macOS.  I also notice that these operations can take _forever_.

Frisch12, 2025-01-31:
Fore me the software constantly crashes when hitting undo or arrange. In arrange it goes to 98% and then crashes.

rkarlsba, 2025-01-31:
So, it's obviouosly a problem in the application itself, or so it seems. Can someone look into this, please? It's of little use if the slicer has multiplate support, but can't use it efficiently.

TargiX, 2025-03-15:
Yeah, and what's even more interesting is that everything works flawlessly in Bambu Slicer. I thought Orca Slicer was supposed to be superior in every way!

TargiX, 2025-03-15:
# Analysis of Slow Undo Operations in OrcaSlicer for Large Projects
## Root Cause Analysis

After examining the codebase, I've identified several potential causes for the slow undo operations:

1. **Complete Model Serialization/Deserialization**: The undo/redo system in OrcaSlicer works by taking complete snapshots of the entire model state. When an undo operation is performed, it deserializes the entire model from the previous snapshot, regardless of how small the change was.

2. **Memory Management Issues**: The `UndoRedo::Stack` class has a memory limit and tries to release least recently used snapshots, but for large projects, even a single snapshot can be very memory-intensive.

3. **Inefficient Snapshot Handling**: The code in `undo_redo_to()` performs many operations when restoring a snapshot, including:
   - Checking for printer technology changes
   - Updating wipe tower configurations
   - Updating UI elements
   - Reloading presets
   - Updating the 3D view

4. **No Incremental/Differential Snapshots**: The current implementation doesn't use incremental or differential snapshots, which would only store the changes between states rather than the entire state.

5. **Lack of Optimization for Large Projects**: There's no special handling for large projects with many objects - the same serialization/deserialization approach is used regardless of project size.

## Potential Solutions

Based on the code analysis, here are potential solutions to improve the performance of undo operations:

1. **Implement Differential Snapshots**: Instead of storing the entire model state, store only the differences between states. This would significantly reduce memory usage and improve performance.

2. **Optimize Serialization/Deserialization**: Improve the efficiency of the serialization and deserialization process, possibly by using a more efficient binary format or by implementing partial serialization.

3. **Lazy Loading**: Implement lazy loading of model components during deserialization, so that only the necessary parts of the model are loaded when needed.

4. **Batch Processing**: Group small changes together to reduce the number of snapshots taken.

5. **Parallel Processing**: Use parallel processing for serialization and deserialization operations to take advantage of multi-core processors.

6. **Memory Optimization**: Improve memory management to reduce memory fragmentation and optimize memory usage.

7. **Profile and Optimize Critical Paths**: Use profiling tools to identify the specific bottlenecks in the undo/redo process and optimize those areas.

8. **Selective Snapshot Taking**: Be more selective about when to take snapshots, possibly avoiding taking snapshots for trivial operations.

## Implementation Recommendation

The most effective approach would likely be to implement differential snapshots, where only the changes between states are stored rather than the entire model state. This would require:

1. Modifying the `UndoRedo::Stack` class to support differential snapshots
2. Implementing a mechanism to track changes to the model
3. Updating the serialization/deserialization process to handle differential snapshots
4. Modifying the `take_snapshot()` and `undo_redo_to()` methods to work with differential snapshots

This would be a significant change to the codebase, but it would likely result in a substantial performance improvement for undo operations, especially for large projects.

## Conclusion

The slow undo operations in OrcaSlicer for large projects are primarily due to the inefficient snapshot-based undo/redo system that serializes and deserializes the entire model state. Implementing differential snapshots and optimizing the serialization/deserialization process would likely result in significant performance improvements.

closed 2025-06-21 by github-actions[bot]