OrcaSlicer/OrcaSlicer #6919: support issue

bayc3d, 2024-09-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 and 2.2.0-beta

### Operating System (OS)

Windows

### OS Version

windows10

### Additional system information

CPU: 10900K RAM 32GB GRAPHICS CARD2070SUPER  3TB SSD

### Printer

2 PIECES PRUSA MK2/3 DIY CLONE

### How to reproduce

I downloaded the parts from https://www.thingiverse.com/thing:3020026. I wanted to print spool holder for diy clone prusa mk2 printer. While printing the file named “3DPNFilHoldRoller.stl” in the zip file I downloaded, I positioned it vertically and activated the support. 
[image]

### Actual results

Whichever support style I choose, the support expands upwards on the lower protrusion where the 608 bearing is to be installed. For this reason, the 608 bearing was printed in such a way that it could not be installed. I only noticed this when I was removing the supports after printing. 
[image]

### Expected results

No matter what I try. Since the result didn't change, I decided to report it here. THIS could be a bug or I could be missing something. But I have tried many times.
[image]
![Ekran Alıntısı3](https://github.com/user-att
[image]
[image]
achments/assets/4c461ed4-0fa7-4430-aed4-49327149ada8)
[image]
[image]

### Project file & Debug log uploads

[3DPNFilHoldGuide.zip](https://github.com/user-attachments/files/17154520/3DPNFilHoldGuide.zip)
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

discip, 2024-09-26:
@bayc3d

Just deactivate **`Make overhangs printable`**. 😊

[image]

bayc3d, 2024-09-28:
Thank you very much. I thought I turned off “make overhang printable”. in fact, I'm sure I did. then I saw that every time I closed and opened the project, it was also selected. problem solved. thanks to everyone who contributed.

closed 2024-09-28 by bayc3d