OrcaSlicer/OrcaSlicer #6923: Travel Exclusion Zones

ToastedMice, 2024-09-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Filament strings when travelling.
This causes problems when inserting bearings in tight tolerances for them to get stuck on those strings. Especially in hard to reach areas where cleaning those strings out is impossible.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Similar to support enforcers/blockers there should be a travel blocker so the print head doesn't enter this area whilst travelling.

discip, 2024-09-27:
@ToastedMice
Have you tried **`Avoid crossing walls`**? 😊

[image]

ToastedMice, 2024-09-27:
Yeah, thats helps when the different sections have walls connecting them but when there aren't, the strings still clog up the parts where the bearing sits. It would be useful for the travels to move along one edge to each section even if it means longer travels.

closed 2025-01-04 by github-actions[bot]