OrcaSlicer/OrcaSlicer #6924: Add back "Support critical regions only" for organic tree supports (feature was deactivated due to bug #2195)

catmag-spec, 2024-09-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Option "Support critical regions only" was removed for organic tree supports due to bug #2195. The underlying bug was not fixed but just the feature/option removed.

Bambu Studio 1.9.7.52 has this option and it is working.

Example from Bambu Studio on:
[image]

Example from Bambu Studio off:
[image]

Missing option in OrcaSlicer:
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add back option "Support critical regions only" for organic tree supports and ensure the option is working.

TransparentHouseSF, 2024-10-12:
same issue.  I have Orca 2.2.2RC and dont see it "Support critical regions only". Its working like a charm in BambuStudio and saving tons of time and material. Why Orca dosnt have such a great option???

closed 2025-01-19 by github-actions[bot]

adobni, 2025-01-26:
Any plans to restore this feature?

lucky13820, 2025-03-22:
I opened a 3MF from Bambu Lab Studio that has this feature enabled, it works in Orca Slicer as well. It's just there's no way to enable this feature in Orca Slicer.