OrcaSlicer/OrcaSlicer #6927: Drawing area not visible - FedoraLinux/Wayland/[Hyprland/Gnome]

SHU-red, 2024-09-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 beta

### Operating System (OS)

Linux

### OS Version

Fedora Workstation 40

### Additional system information

```
             .',;::::;,'.                 shured@fedora
         .';:cccccccccccc:;,.             -------------
      .;cccccccccccccccccccccc;.          OS: Fedora Linux 40 (Workstation Edition) x86_64
    .:cccccccccccccccccccccccccc:.        Host: MS-7E27 (1.0)
  .;ccccccccccccc;.:dddl:.;ccccccc;.      Kernel: Linux 6.10.11-200.fc40.x86_64
 .:ccccccccccccc;OWMKOOXMWd;ccccccc:.     Uptime: 43 mins
.:ccccccccccccc;KMMc;cc;xMMc;ccccccc:.    Packages: 2555 (rpm), 9 (flatpak), 39 (brew)
,cccccccccccccc;MMM.;cc;;WW:;cccccccc,    Shell: fish 3.7.0
:cccccccccccccc;MMM.;cccccccccccccccc:    Display (LG ULTRAWIDE): 3440x1440 @ 50 Hz in 34″ ]
:ccccccc;oxOOOo;MMM000k.;cccccccccccc:    DE: sway
cccccc;0MMKxdd:;MMMkddc.;cccccccccccc;    WM: Hyprland (Wayland)
ccccc;XMO';cccc;MMM.;cccccccccccccccc'    Cursor: Adwaita
ccccc;MMo;ccccc;MMW.;ccccccccccccccc;     Terminal: kitty 0.36.2
ccccc;0MNc.ccc.xMMd;ccccccccccccccc;      Terminal Font: FiraCode-Regular (10.5pt)
cccccc;dNMWXXXWM0:;cccccccccccccc:,       CPU: AMD Ryzen 5 7600X (12) @ 5.45 GHz
cccccccc;.:odl:.;cccccccccccccc:,.        GPU 1: NVIDIA GeForce RTX 3050 6GB [Discrete]
ccccccccccccccccccccccccccccc:'.          GPU 2: AMD Raphael [Integrated]
:ccccccccccccccccccccccc:;,..             Memory: 7.37 GiB / 30.45 GiB (24%)
 ':cccccccccccccccc::;,.                  Swap: 0 B / 8.00 GiB (0%)
                                          Disk (/): 157.57 GiB / 1.82 TiB (8%) - btrfs
                                          Local IP (enp12s0): 192.168.178.20/24
                                          Locale: en_US.UTF-8
```

### Printer

Ender 3

### How to reproduce

1. Open Orca
2. 3D area not visible

- Orca is used as AppImage
- It worked before but now it stopped working
- Also starting an older image does not retrieve the error
- It does not work on Hyprland and GNOME (with Wayland)

[image]

### Actual results

3D area is not showing anything

### Expected results

3D area is shown

### Project file & Debug log uploads

```
[... 294 lines cut ...]

(orca-slicer:106291): Gtk-CRITICAL **: 12:43:43.022: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:106291): Gtk-CRITICAL **: 12:43:43.022: gtk_widget_set_size_request: assertion 'width >= -1' failed
Failed to create GBM buffer of size 836x1316: Invalid argument

(orca-slicer:106291): Gtk-CRITICAL **: 12:43:43.601: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:106291): Gtk-CRITICAL **: 12:43:43.602: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:106291): Gtk-CRITICAL **: 12:43:43.604: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:106291): Gtk-CRITICAL **: 12:43:43.604: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:106291): Gtk-CRITICAL **: 12:43:43.605: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:106291): Gtk-CRITICAL **: 12:43:43.606: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:106291): Gtk-CRITICAL **: 12:43:43.606: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
```

### Checklist of files to include

- [X] Log file
- [ ] Project file

Noisyfox, 2024-09-27:
https://github.com/bambulab/BambuStudio/issues/4669

Are you using the same Nvidia GPU driver as mentioned in this thread too?

SHU-red, 2024-09-27:
Yes sir

```
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 560.35.03              Driver Version: 560.35.03      CUDA Version: 12.6     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3050        Off |   00000000:01:00.0  On |                  N/A |
| 38%   48C    P0             22W /   70W |     734MiB /   6144MiB |      3%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A    113005      G   Hyprland                                      238MiB |
|    0   N/A  N/A    113271      G   kitty                                          91MiB |
|    0   N/A  N/A    113458      G   Xwayland                                        8MiB |
|    0   N/A  N/A    282581      G   zen                                           231MiB |
|    0   N/A  N/A    283734      G   kitty                                          43MiB |
+-----------------------------------------------------------------------------------------+
```

I see, that changing ENVVARS could help
Thank you for taking the time

SHU-red, 2024-09-27:
Yes, this is the command
```
env __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_DISABLE_DMABUF_RENDERER=1 WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 ~/AppImages/gearlever_orcaslicer_*
```
Thank you very much!

closed 2024-09-27 by SHU-red