OrcaSlicer/OrcaSlicer #6936: X.local links for klipper printers wont upload or print from the ui

hYdos, 2024-09-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

latest

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

N/A

### Printer

Any Klipper Printer

### How to reproduce

1. use a klipper printer
2. set its ip as raspberrypi.local (test confirms its working fine)
3. try to use upload and print and watch as it says completed even when it failed

### Actual results

  See Above.

### Expected results

See Above.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I will be surprised if this is even looked at with the disregard ive seen here.

tomoinn, 2024-09-30:
This works fine for me, running from win11, linux and macos. Win10 specific?

hYdos, 2024-09-30:
Possibly. Setting the ip directly works fine

closed 2025-01-06 by github-actions[bot]