OrcaSlicer/OrcaSlicer #6937: Paint ironing

rbuckton, 2024-09-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

On some models I only want ironing enabled on specific surfaces. The only way to accomplish this today is with modifiers, but this is fairly tedious.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Much like color painting or seam painting, it would be far easier to be able to paint ironing onto specific surfaces

### Describe alternatives you've considered

1. Height range modifiers - works but tedious. requires jumping back and forth between slicing/preview to determine appropriate layer heights.
2. Object modifiers - works but tedious. requires manually positioning the object in the correct locations to enable/disable ironing.

closed 2025-01-04 by github-actions[bot]

mysterious-roman-dodecahedron, 2025-08-15:
Agree with scrith. Layer range modifiers are clunky to set up for this, and using a modifier object to create a zone that allows ironing has created some strange artifacts with walls and top layers when I tried in the past. I think it would be an immensely useful feature

Kingsidorak, 2025-08-24:
I need ironing on specific parts of several models that I have to keep updating, so this would be far better than those stupid added in blocks and whatever with different settings