OrcaSlicer/OrcaSlicer #6941: "Bridge counterbore holes" set to "partially Bridged" leads to strange overhang behaviour (bridge instead of overhang)

imaluschi, 2024-09-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta2

### Operating System (OS)

Linux

### OS Version

Fedora 40

### Printer

Bambu Lab X1 Carbon

### How to reproduce

(didn't try with other parts!)
* Two similar parts with overhang, one was changed due to functionality with a small pad under an overhang (see pictures).
* In my used settings "Bridge counterbore holes" is set to "partially Bridged". No holes in the part so I didn't care.
* Slice
* See the different line types on layer 44 @ 8,85 mm - be aware of the small pad on the right side of the bridge. This is the only difference between the two models.
[image]

### Actual results

In the first layer with overhang (layer 44) the overhang lines were made as overhangs. The part came out nice and were usable "out of the printer".
A little change due to functionality changed the overhang lines in layer 44 to "Bridge" instead of "Overhang wall". Some of the printed lines were hanging down so finishing with a knife was mandatory.

### Expected results

In two nearly exakt identical parts overhangs should be treated on exactly the same way.

### Project file & Debug log uploads

Project file
[2024-09-27_Parts_strange_overhang_behaviour_Bridge_counterbore_holes.zip](https://github.com/user-attachments/files/17169700/2024-09-27_Parts_strange_overhang_behaviour_Bridge_counterbore_holes.zip)

Log-file
[debug_Fri_Sep_27_22_35_24_7084.log.0.zip](https://github.com/user-attachments/files/17169717/debug_Fri_Sep_27_22_35_24_7084.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

"Bridge counterbore holes" set to "partially Bridged"
[image]

"Bridge counterbore holes" set to "None". Only this setting changed!
[image]

imaluschi, 2024-09-27:
[image]

asphaltbuffet, 2024-11-20:
I believe i'm getting the same behavior in the Windows application (2.2.0).

With "Bridge counterbore holes" set to "partially bridged" and using the built-in orca cube:
[image]

And then with it set to "none":
[image]

staal54a, 2024-12-16:
What's interesting about this is setting to "partially Bridged" actually allows me to use actual bridging settings for an entire bridge (otherwise at least part of it is printed with overhang settings). I don't know if it's just Orca or one of the upstream slicers like Bambu Studio or PrusaSlicer that introduced it, but there really are some problems with overhangs and bridging clobbering each other.

closed 2025-03-25 by github-actions[bot]