OrcaSlicer/OrcaSlicer #6945: Add profile for Qidi Plus 4

biged83, 2024-09-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

no

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Add a good working profile for qidi plus 4

Qjeezy, 2024-09-29:
100% agree. The QIDI slicer looks the same but has many differences Im not used to. I'd prefer to keep using orca slicer.

kzemlin, 2024-11-02:
Another vote.  I'm not enough of a software guy to know what I'm doing.  I tried copying all machine settings and machine G-Codes from the QIDI slicer into the generic KLIPPER profile, but I can't slice.

Failed to generate G-code for invalid custom G-code.
machine_start_gcode Parsing error at line 1: Variable does not exist
PRINT_START BED=[bed_temperature_initial_layer_single] HOTEND=[nozzle_temperature_initial_layer]
CHAMBER=[chamber_temperatures]

The one particular setting I rely on for clean prints is extra length on restart.  The QIDI slicer does not have that and I haven't figured out a work-around.

Qjeezy, 2024-11-02:
> Another vote.  I'm not enough of a software guy to know what I'm doing.  I tried copying all machine settings and machine G-Codes from the QIDI slicer into the generic KLIPPER profile, but I can't [...]

There are plus 4 profiles now

kzemlin, 2024-11-03:
Updated - running a test print now.  Thank you!

closed 2025-02-09 by github-actions[bot]