OrcaSlicer/OrcaSlicer #6949: Arachne leaves print area

EXOgreen, 2024-09-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu P1S

### How to reproduce

1. Switch Classic Generator to Arachne Generator
2. slice file

### Actual results

Phantom lines extend off to infinity.
[image]
[image]

### Expected results

No phantom lines are created.
[image]

### Project file & Debug log uploads

[Block52 3D Print ARACHNE-ERROR.zip](https://github.com/user-attachments/files/17172934/Block52.3D.Print.ARACHNE-ERROR.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

This does not happen in 
- PrusaSlicer 2.8.1
- Bambu Studio 1.9.7.52

tested in OrcaSlicer 2.2.0-beta and 2.1.1

When switching wall generator from classic to Aranche, I get (in this one model) phantom lines that extend out of the printers build volume. In this example, the inner wall starts printing, then around line 387598 turns around, prints to X3624.45, and then comes back to the model and meets with the start of its inner wall.

Noisyfox, 2024-09-28:
Not sure why PS and BBS don't have this issue, but you could try fixing the model first.

EXOgreen, 2024-10-02:
Using Fix Model does make the issue go away. I was more concerned with why it only affects OrcaSlicer.

Noisyfox, 2024-11-01:
https://github.com/SoftFever/OrcaSlicer/pull/7330 should fix the issue:
[image]

closed 2024-12-18 by SoftFever

EXOgreen, 2024-12-18:
You're awesome! Thanks!

PR #7330 (2024-12-18): Arachne: Fix an arithmetic overflow that causing extra points outside plate
Fix #6949
Fix #7321

[image]