OrcaSlicer/OrcaSlicer #6954: Penetrating colors into adjacent layers

SalatDressingOil, 2024-09-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.1.1 и V2.2.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu lab X1

### How to reproduce

I'm not sure if this is a bug, but I can't consider it the right behavior. Let's say we have a standard cube, when marking colors using a standard tool, I select red and a range of layers, set 0.10, and paint by clicking on the bottom surface of the cube.
[image]

### Actual results

 next, when slicing:
[image]
[image]
[image]
the red layers penetrate into the white ones, which is why the printer needs to change color once again, wasting plastic and time. The height of my hair is 0.1 mm.

### Expected results

 I expect that with this approach, only the first layer will be colored. whatever the problem is, it really interferes with color slicing, which is why I'm starting to avoid it..

### Project file & Debug log uploads

[Куб.zip](https://github.com/user-attachments/files/17174585/default.zip)
[debug_Sat_Sep_28_13_35_50_32532.log.zip](https://github.com/user-attachments/files/17174586/debug_Sat_Sep_28_13_35_50_32532.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

we just tested it in Bombustudio, it seems to be their problem.. I'm not sure, but whatever it is, it needs to be solved anyway, right?
and yes, I use a translator from Russian, I'm sorry if something is unclear

Noisyfox, 2024-09-29:
If you need to color a single layer, consider using a modifier. The paint tool does not gaurantee how much the painted color will go into the object, and some people want the color to penetrate more (such as for semi-transparent filament).
Maybe we could introduce some options to control how deep the painted color should go under the surface.

SalatDressingOil, 2024-09-29:
> If you need to color a single layer, consider using a modifier. The paint tool does not gaurantee how much the painted color will go into the object, and some people want the color to penetrate more [...]

it really happens to be useful when one plastic penetrates into the layers of another, it can probably also increase strength, but for me it often leads to unnecessary mixing of colors, for example, against the background of several upper white layers, the lower ones shine through with a brighter color, given that with complicated geometry, penetration is quite chaotic - it makes unpleasant smears in enlightenment. This is just one of the many examples I've come across... so it would be really helpful.

SalatDressingOil, 2024-09-29:
> If you need to color a single layer, consider using a modifier.

Could you tell us in more detail what it is?

closed 2025-01-06 by github-actions[bot]