OrcaSlicer/OrcaSlicer #6957: Please add a Creality CR-M4 profile

Octipole, 2024-09-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Please provide a printer profile for the Creality CR-M4 
preferably for large nozzles 0.6, 0.8, and 1.0mm 
A large printer needs large nozzles.

cochcoder, 2024-10-07:
@Octipole 
I wasn't able to make a profile for the larger nozzle sizes for the CR-M4 as there was nothing that I could base it off of. However, I did create a profile for the 0.4 nozzle, if you would like to make a profile based off that for larger nozzle sizes, I'd be more than happy to add them into Orca Slicer on your behalf.

closed 2024-10-07 by SoftFever

Octipole, 2024-10-14:
Thanks 
Not sure I'm posting in the right spot. I just deleted line 7 to start trying the profile (prasing error line 7 m109
[image]

cochcoder, 2024-10-14:
> Thanks Not sure I'm posting in the right spot. I just deleted line 7 to start trying the profile (prasing error line 7 m109 ![line 7](https://private-user-images.githubusercontent.com/163933524/376099894-23723f01-b573-4d77-81c4-0748962603fc.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg5MTkwNDIsIm5iZiI6MTcyODkxODc0MiwicGF0aCI6Ii8xNjM5MzM1MjQvMzc2MDk5ODk0LTIzNzIzZjAxLWI1NzMtNGQ3Ny04MWM0LTA3NDg5NjI2MDNmYy5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAxNFQxNTEyMjJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYzAxZTZmNzNlMGFkZjQyNDBkMDA3MDZhZDY0MGU1NGEwNDQxODNmMDZhNWM5MjdjNGU2MjI5OTg2NzllNzViJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.HCcVwDwFe7YdPANID0h47W5BtdJbd70tI_H3_sKpmUU)

Thanks for finding this! If you change the following values, would it work? 

`max acceleration extruding: 700`

```
start G-Code: 
    M220 S100 ;Reset Feedrate
    M221 S100 ;Reset Flowrate
    M140 S[bed_temperature_initial_layer_single] ; set final bed temp
    G28 ;Home
    G92 E0 ;Reset Extruder
    G1 Z2.0 F3000 ;Move Z Axis up
    G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
    M104 S[nozzle_temperature_initial_layer] ; set final nozzle temp
    M190 S[bed_temperature_initial_layer_single] ; wait for bed temp to stabilize
    M109 S[nozzle_temperature_initial_layer] ; wait for nozzle temp to stabilize
    G1 X10.1 Y145.0 Z0.28 F1500.0 E15 ;Draw the first line
    G1 X10.4 Y145.0 Z0.28 F5000.0 ;Move to side a little
    G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
    G92 E0 ;Reset Extruder
    G1 E-1.0000 F1800 ;Retract a bit
    G1 Z2.0 F3000 ;Move Z Axis up
    G1 E0.0000 F1800
```

Octipole, 2024-10-17:
Thanks for the follow up. I started a 4 day print yesterday 10/16 I should of mentioned I'm using a creality sonic pad. It's glitchy. After struggling with larger nozzles, which would lessen the time when something could go wrong I'm back using 0.4 I might try a 0.6mm as a compromise. I'll try the new start g code when I start the next print. Thanks again

PR #7002 (2024-10-07): Add Creality K1 SE, K2 Plus, & CR-M4
Closes #6984, closes #6707, closes #6957
commit 01ffeb6b (2024-10-07): Add Creality K1 SE, K2 Plus, & CR-M4 (#7002)

Closes #6984, closes #6707, closes #6957