OrcaSlicer/OrcaSlicer #6958: Bridge Flow Ratio Is Changing Infill Spacing

CCS86, 2024-09-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2 Beta 2

### Operating System (OS)

Windows

### OS Version

10 x64

### Printer

X1C

### How to reproduce

...

### Actual results

...

### Expected results

When using "Bridge Flow Ratio" to tune flow over support, I would expect only the the extrusion amount for each line to change. Looking at perimeter walls, this seems to be the case. But, the infill lines are having their spacing altered, which is really just changing line width, now specifically flow:

[image]

### Project file & Debug log uploads

...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

julianjc84, 2024-10-15:
Adding to this issue:
Is Line Type: Internal Bridge. Supposed to be impacted by the Bridging: Bridge Density % As when I adjust the Bridge Density% I see no changes on Line Type: Internal Bridge.

https://github.com/user-attachments/assets/162157b6-27d2-4b13-8fd7-2b69cebfc741

igiannakas, 2025-01-31:
That is by design as lower flow -> thinner lines -> therefore the lines need to be closer together to create the same surface effect.

I am about to raise a PR with some additional options for internal bridges, including control over the spacing.

CCS86, 2025-02-01:
> That is by design as lower flow -> thinner lines -> therefore the lines need to be closer together to create the same surface effect.
> 
> I am about to raise a PR with some additional options for internal bridges, including control over the spacing.

Ah, that makes sense.

igiannakas, 2025-02-01:
pr is raised btw. Take a look here: https://github.com/SoftFever/OrcaSlicer/pull/8263

More to come but that’s a start :)

closed 2025-02-12 by SoftFever