OrcaSlicer/OrcaSlicer #6961: Extrusion move VERY far outside of build plate

ksummers92, 2024-09-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Switchwire

### How to reproduce

Slice a model with single-extruder multimaterial using multiple filaments in a single print where the color changes start at the build plate.

### Actual results

Random layer produces an extrusion move to Y11957.445. Does not occur in a layer that contains any tool changes. 

[image]

Something similar happens when the extrusion occurs in a layer which a toolchange occurs resulting in unnecessarily large purge tower.
https://discord.com/channels/1137181739773603922/1278158595615363146

### Expected results

An extrusion to 12 meters should not be produced.

### Project file & Debug log uploads

[Tip_Catcher_MMU.zip](https://github.com/user-attachments/files/17181069/Tip_Catcher_MMU.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

This seems to only occur when slicing a multimaterial print that has a toolchange on the first layer. I have sliced another multimaterial print with a toolchange later in the print and did not have this issue.

yonkiman, 2024-12-03:
I have a similar bug.  OrcaSlicer 2.2.0, but I'm not using multimaterial.  Gcode tries to move 8 meters on a 120mm bed.

In the attached gcode, the command for line number 151841 is `G1 X55.506 Y8403.722 E229.77609`.  Part sliced fine in SuperSlicer and Simplify3D, only appears in OrcaSlicer.  I had to deleted the gcode after the error to make the zip file less than 1MB.  But loading the attached 3mf and slicing should recreate the problem.

[Bad gcode on line 151841 trimmed.zip](https://github.com/user-attachments/files/17998519/Bad.gcode.on.line.151841.trimmed.zip)

[Dragon Burner 8 meter Y move.zip](https://github.com/user-attachments/files/17998577/Dragon.Burner.8.meter.Y.move.zip)

ndevenish, 2025-01-04:
Same issue as https://github.com/SoftFever/OrcaSlicer/pull/7330 ?

ksummers92, 2025-01-05:
Could be. I never tried slicing without Arachne to see if it changed. Seems like it was fixed in the latest update? I'll try it again.

closed 2025-04-13 by github-actions[bot]