OrcaSlicer/OrcaSlicer #6971: part cooling fan speed control settings

SarraceniaWilds, 2024-09-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Have feature specific fan control speeds been added? This has been requested a few times but autoclosed and no action taken as far as I can tell and I dont see any option to reopen the issue. re: https://github.com/SoftFever/OrcaSlicer/issues/1972 or #5328? #4370?

> ![screenshot](https://private-user-images.githubusercontent.com/130192690/344501213-dfff78d2-b644-4d50-bef1-401539efeb8c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc3MTgzMzgsIm5iZiI6MTcyNzcxODAzOCwicGF0aCI6Ii8xMzAxOTI2OTAvMzQ0NTAxMjEzLWRmZmY3OGQyLWI2NDQtNGQ1MC1iZWYxLTQwMTUzOWVmZWI4Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkzMFQxNzQwMzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03MmY2ODkyMTIxZDE1ZmUzOWJlZWZkYjFjZTMxZDljNGE4OWNkODVhM2UxOTI1NDg5NmU0NGViYmRmMzNhOWU4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.zZYP_8wv2IUyvUn6nW4C0lddiaOJGEEUS5t0y62m8D0) Screenshot to illustrate request. Lack of feature specific control for fan, is there an open issue for this request?

> ![image](https://private-user-images.githubusercontent.com/68181325/265176893-7d4c4cf9-c115-4d0d-a8d8-759b628ffeaf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc3MTgzMzgsIm5iZiI6MTcyNzcxODAzOCwicGF0aCI6Ii82ODE4MTMyNS8yNjUxNzY4OTMtN2Q0YzRjZjktYzExNS00ZDBkLWE4ZDgtNzU5YjYyOGZmZWFmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTMwVDE3NDAzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFhMDdlNWI4N2QxMGE1N2I4M2I5Y2IwZTI1MTNjNzZjMWY3YjFlYTI4ZWQ0MzcxMWFkY2JkNjU5MDg0ODllZmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.yGmzz5aEe1_wHgAmOsUzvEuZsz0OkBoQ8kWcBjlnGbM)
> 
> could we please get the settings for fan speed control, like superslicer has ?
> 
> I was going to try orcaslicer on my klipper printer but noticed it doesnt have any fine tuning controls for part cooling fan.
> 
> i have included a screenshot of the settings that superslicer has.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Please add feature specific fan speed controls.

### Describe alternatives you've considered

The current alternatives are either use a different slicer or suffer low quality prints.

macjoel1991, 2024-09-30:
I second this

discip, 2024-10-01:
@SarraceniaWilds
Does this PR #6738 cover what you are looking for? 😊

SarraceniaWilds, 2024-10-01:
@discip 

No but its a start

closed 2025-01-07 by github-actions[bot]

SarraceniaWilds, 2025-01-17:
Bump. Still an issue. Unresolved in any way.