OrcaSlicer/OrcaSlicer #6976: Multiple dovetail cuts

liftbag, 2024-09-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I recently helped a friend perform the cutting of an object with two dovetails, using a complicated procedure of cuts and boolean operations, as the current dovetail cut function does not allow for the insertion of more than one dovetail.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I was thinking of implementing the option to insert at least a second dovetail, with its parameters, or even as a child of the first one, but with the ability to be positioned freely along the cutting axis.

### Describe alternatives you've considered

Multiple cut and sew operations only.

### Additional context

[image]

[image]

[image]

paulohbsa, 2024-10-28:
I think the same.
Actually, I was a little bit frustrated when I realized it was not possible to do multiple dovetail cuts.

cai876, 2024-12-24:
I agree, I would really like this feature too. One dovetail in not suitable for large parts

pepies2137, 2025-02-23:
Found it searching for that function on google. Still a request guys. It would be really helpful :)

dcchillin46, 2025-04-21:
Ya this would be handy

RobertKoval, 2025-06-14:
up! it's weird that this feature wasn't implemented from the start.

carlinogianluca, 2025-11-24:
+1 ia a must have feuture

Kasperinac, 2025-12-19:
This is exact what i need

ezric1, 2026-01-01:
Seems to me that instances along a plane should doable, similar to how Fusion operates. I would love this feature!

liftbag, 2026-01-04:
I have the impression that an issue that has been open for 15 months is practically invisible to the developers. Between the bot that closes issues due to inactivity and the sheer number of issues opened every day, I can understand how this happens.
I was wondering if there is an alternative channel, possibly connected to GitHub, to help bring feature requests or long-standing issues to the developers’ attention.

Felix14-v2, 2026-08-31:
@RF47 implemented, I guess?

RF47, 2026-08-31:
Implemented in #12318

closed 2026-08-31 by RF47