OrcaSlicer/OrcaSlicer #6978: temp tower - offer scaling to suit large nozzles

derFliegendeHamburger, 2024-10-01:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I use 0.8mm nozzle on this printer with 0.4mm up to 0.5mm layer height. The offered calibration/temp tower does not take this into account but has a fixed size which I find too small for this larger nozzle. It results in (too) few layers, barely readable text and (too) short cooling time between layers. The size is too small for the typical application with larger nozzles and the resulting temp tower is of little value.

Manually scaling the temp tower in Z-direction is not possible or it will also scale the temp range beyond the desired range!

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It should allow to manually or automatically adapt the temp tower size to layer width and height

### Describe alternatives you've considered

1. manually scale XY - works
2. manually scale Z - side effect : breaks the temp range beyond the desired scope

rmeissn, 2024-10-08:
The same issue also applies to the retraction tower.

I attempted to fine-tune the retraction settings for wood filament using a 0.8 mm nozzle. However, the process would not accommodate a 0.4 mm layer height and instead defaulted to a 0.2 mm layer height, particularly with a very small model in terms of height. The results were difficult to interpret...

Adapting calibration models to the nozzle width sounds like a reasonable path, e.g. by scaling these or using different models for different nozzle sizes.

rmeissn, 2025-01-07:
Still no change since 2.2.0

closed 2026-04-26 by SoftFever