OrcaSlicer/OrcaSlicer #6985: More Accurate Retraction Calibration Tower

MxBrnr, 2024-10-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Current retraction calibration method is time and material consuming, with less accurate and generally difficult-to-interpret results than other options.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Replace the current single-function retraction calibration model with a combined retraction distance/speed tower similar to that found at http://www.retractioncalibration.com/

### Additional context

This allows the user to test and see the close relationship between retraction distance and speed, and to compare the combined results side-by-side to that of other pairs of retraction settings, all done in a single print test model.

Ergonomicmike, 2024-10-02:
My 2 cents: I've printed the retractioncalibration test in the past. (For fun.) I can't tell you why it is, but I've found that the "normal" retraction tower test gives much more real world results. (And this is on a Bowden printer, the worst case for stringing.)

That is, I can use lower retraction settings (always a good thing) based on a retraction tower test to eliminate stringing, as opposed to the higher settings based from retractioncalibration.

It might be that the many variables involved in retraction nowadays (mainly PA, but perhaps others, like wiping?) aren't triggered by the retractioncalibration test?

I dunno. But I know empirically that the retractioncalibration test is overkill for me. (Albeit fascinating.)

And while speed of retraction makes some difference, it seems to be a 2nd order effect, retraction distance making the most difference.

P.S. I use a loop to view the results of a retraction tower test because there are some subtleties that can only be seen upon magnification. But if the primary goal of using a retraction test is to eliminate stringing, then, for me, the tower test meets that goal.

Ergonomicmike, 2024-10-03:
Interesting that we have opposite experiences. 

My main point, as you acknowledged, is that the retractioncalibration test is (I believe) hardcode gcode, bypassing any enhancements provided by Orca or the printer's firmware. (e.g. Klipper.) 

So while it's true that if one can print the retractioncalibration test without stringing, then one should not have stringing when the slicer and firmware are involved. But it might not be necessary to use, say, 5 mm of retraction (overkill) to pass the test when 2.4 mm retraction will work fine when the slicer and firmware are involved.

But a model like the retractioncalbiration test, sliced in Orca, might be interesting. (Althoug I wonder how one forces the extremes of the test via an stl.)