OrcaSlicer/OrcaSlicer #6987: data_dir and opening associated files

XEGARE, 2024-10-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta2

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Flying Bear Ghost 6

### How to reproduce

1. Use data_dir
2. Set up associations
3. Open stl/step/3mf file.

### Actual results

data_dir is not used, the OrcaSlicer folder is created in AppData and the initial application configuration window appears

### Expected results

data_dir should be used

### Project file & Debug log uploads

No logs

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DonutCables, 2024-10-02:
I'm encountering the same issue when sending a file to Orca from the 3DPrintPlus plugin in Fusion. It seems like opening Orca indirectly is causing it to not look for `data_dir`.

SoftFever, 2024-10-08:
should be fixed now

PR #7049 (2024-10-08): Fixed an issue that if Orca was invoke by associated files, local data_dir was not used
# Description

fixed #6987
# Screenshots/Recordings/Graphs

## Tests
commit fb032cd8 (2024-10-08): Fixed an issue that if Orca was invoke by associated files, local data_dir was not used (#7049)

# Description

fixed #6987
# Screenshots/Recordings/Graphs

## Tests
closed 2024-10-08 by SoftFever