OrcaSlicer/OrcaSlicer #6989: YOLO Test doesnt work?

SphaeroX, 2024-10-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Sovol SV08

### How to reproduce

load the YOLO test

### Actual results

[Orca YOLO Flow Calibration.zip](https://github.com/user-attachments/files/17232559/Orca.YOLO.Flow.Calibration.zip)
[image]
[image]

The E at the GCode wont change, also i never feel any differnce between the tests.

### Expected results

Extrusions should be differnt at GCode for the Objetcs

### Project file & Debug log uploads

[flowrate_0_PETG_0.3_20m58s.zip](https://github.com/user-attachments/files/17232599/flowrate_0_PETG_0.3_20m58s.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tlhintoq, 2024-10-13:
I thought it was just me after running this test on my X1C with .6 nozzle.
I couldn't detect ANY difference in any of the chips.

SphaeroX, 2024-10-13:
I also got a 0.6 Nozzle 🤔

tlhintoq, 2024-11-16:
Update:  The problem is first layer thickness.

Print the first layer at 0.20 or thicker  and the size tab doesn't print.
[image]

Slice it with first later a 0.16 and it does slice with a printable size tab.
[image]

SphaeroX, 2024-11-16:
Thank you very much for showing the solution and also respect for figuring it out! Hopefully, this will be fixed at some point. @SoftFever

tlhintoq, 2024-11-19:
@SoftFever   
While you're looking at this... The test doesn't re-space the chips when it automatically resizes them based on a larger nozzle.

For example, when you go to a 1.0 nozzle
Make the test
Then reposition the test, all the chips will resize larger to accommodate the big nozzle.
BUT they don't re-space - so all the chips overlap.

https://github.com/user-attachments/assets/b83bfc86-68ee-4bcb-a851-1cf03d24181d

[image]

Akucuki, 2025-01-03:
I'm also having some issues with this on 0.6 nozzle. My test tabs print fine except there's no visible difference between their surfaces. Setting the first layer height to 0.16 instead of my default 0.24 didn't help.

Akucuki, 2025-01-04:
I tested it with 0.4 and still got the identical plates printed. The traditional Pass 1/Pass 2 tests are working correctly

snakeRuSE, 2025-01-10:
0.8 nozzle

before
[image]
after
[image]

SoftFever, 2025-02-15:
> [@SoftFever](https://github.com/SoftFever) While you're looking at this... The test doesn't re-space the chips when it automatically resizes them based on a larger nozzle.
> 
> For example, when you go to a 1.0 nozzle Make the test Then reposition the test, all the chips will resize larger to accommodate the big nozzle. BUT they don't re-space - so all the chips overlap.
> 
>  2024-11-19_10-24-50.mp4 
> [image]

Yesh, this is a limitation in all flow rate tests.
Currently, users need to hit the auto arrange button to arrange it unfortunately.

SoftFever, 2025-02-15:
Hey guys,
Just want to double check.
I vaguely remember that I have fixed this issue, I probably forgot to update the status here..
I can't reproduce the bug in 2.2 official or 2.3-dev build.
I fixed a bug for 2.2 official when the flow test won't work on France/German OS(or any language where they don't use . to denote decimal).

Please let me know if you still experience the issue.

mattaw, 2025-02-23:
I've just hit this with an 0.6mm nozzle, I didn't find this issue unfortunately, so created a new issue #8505 . Multiple attempts to print tests, and they all have the same top surface.

SoftFever, 2025-02-23:
@mattaw and all who can experience this issue, please check my last comment.
I can't reproduce the bug. 
I noticed in the project you attached, flow ratio of each part is wrong, they are all set to 1.

[image]

freakydude, 2025-02-27:
I cannot see the problems in 2.3.0-beta2 after slicing the plate - the flow for the individual objects looks correct. Didn't print it.

closed 2025-02-28 by SoftFever