OrcaSlicer/OrcaSlicer #7007: Allow to modify layer height limits as a percentage of noozle diameter

JuanR3D, 2024-10-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Actual layer height limits are expressed as a static value, it doesn't matter the noozle_diameter you configure in your printer.
My experience tells me that the layer height limits are between 20% and 80% of the noozle diameter.
As i want to use my printer with various noozle diameters, in a easy and simple way, i consider important to have these parameters correctly configured.

A solution for this is to create various printer profiles for every noozle i want to use, but i consider this a waste of time, i'd like just to change the noozle diameter in my printer profile and have almost everything configured as a percentage of it so it changes automatically.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Layer height limits must can be expressed as a percentage of noozle diameter.

### Additional context

[image]

ShaneDelmore, 2024-10-05:
I too wish nearly ever setting related to nozzle diameter offered a percentage option.  The few that don't keep me from using a profile for different nozzle sizes.

closed 2025-01-11 by github-actions[bot]

ShaneDelmore, 2025-01-18:
@JuanR3D I looked into implementing this and realized it is more complicated than it appears as you may have multiple extruders, each with different nozzle sizes.  In this case a fixed layer height still works as long as it is a height that can be printed by all of the nozzles involved in the print, but a percent layer height would not work as it would give different layer heights for each nozzle.

JuanR3D, 2025-01-20:
In that case, it would be necessary to check if more than one extruder is going to be used in the print (check filaments used in the piece), so it would generate two different scenarios:

When only one extruder is going to be used, those height limits will apply to its parameters.
When more than one extruder is going to be used, the height limits will be: the lower value of both for the max height, and the higher value for the min height.

For example, let's suppose that we have two extruders. One has a nozzle diameter of 0.4, and the other one has 1.2.
The max height would be set by the 0.4 nozzle, as 0.32 mm (80% of nozzle diameter), and the min height would be set by the 1.2 nozzle, as 0.24 mm (20% of nozzle diameter).

I know that this can be quite difficult to implement, as print parameters shouldn't depend on what is going to be printed (the fact of checking the filaments used in the print), but I don't know, maybe one solution might be to set the height limits according to the principal extruder (T0), and only show a warning when slicing the model if it is detected that more extruders are going to be used in the piece and the layer heights are over their respective limits.