OrcaSlicer/OrcaSlicer #7009: User Preset is not saved

kingnoob1377, 2024-10-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Custom Printer

### How to reproduce

1. make any changes to a preset
2. save the changed preset with a different name
3. close the slicer
4. open the slicer

### Actual results

The saved user preset is gone, and what should be saved has become unsaved changes for the initial preset

### Expected results

User presets should be saved

### Project file & Debug log uploads

[debug_Fri_Oct_04_20_28_47_18696.log (2).zip](https://github.com/user-attachments/files/17257608/debug_Fri_Oct_04_20_28_47_18696.log.2.zip)
[9mm_2WD_XY_joint_left_b.zip](https://github.com/user-attachments/files/17257630/9mm_2WD_XY_joint_left_b.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Trying with different projects yields the same result

slipx06, 2024-10-15:
I can confirm this issue

dmgualb, 2024-10-16:
This happens to me too, using 2.2.0-beta on Windows, but I have additional information: If you close OrcaSlicer again and reopen, the user presets are there! Oddly enough, if you close-reopen, they're gone again... and if you close-reopen, they're back... and so on. So, to use custom settings, I have to open OrcaSlicer and check if my settings are there. If they're not, I just have to close and reopen OrcaSlicer and they'll be back... It's really anoying. Happens to me in all computers I've tried, Windows 11 and Windows 10.

closed 2025-01-23 by github-actions[bot]