OrcaSlicer/OrcaSlicer #7013: When printing "The friendly articulated Slug" (https://www.thingiverse.com/thing:4727448), Orca crashes

The-xyph, 2024-10-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

11

### Additional system information

Intel G14
128Gb
RTX 4090

### Printer

Neptune 4 PRO

### How to reproduce

Download https://www.thingiverse.com/thing:4727448 and slice it.. I had some luck by using "classic" mode instead of Arachne.. 

Slice, and observe Orca disappear without showing an error.. just "poof"!

### Actual results

15% is reached and then "poof"

### Expected results

Model slices ready to print.

### Project file & Debug log uploads

[SlugV1.1.zip](https://github.com/user-attachments/files/17262477/SlugV1.1.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ergonomicmike, 2024-10-04:
Sliced without crashing here. (2.2.0-beta.)

The-xyph, 2024-10-05:
I tried again with 2.2.0-beta and it gets further along in the slicing and then crashes as before.

Ergonomicmike, 2024-10-05:
It sounds like the problem is following your hardware and/or Win11 and/or your slicer settings. 

(I'm using Win10.)

Are you using the default slicer settings for your printer as a vanilla baseline? (You implied that it completed slicing when using Classic wall generator.)

More input is needed from others who try to slice the slug model to know if this problem is unique to your setup or is due to the slicer proper.

The-xyph, 2024-10-05:
I wondered the same thing, perhaps a hardware/setup issue.

I believe I got it to slice with classic, but it doesn't work now
unfortunately.. which is a bit weird tbh!

I'll try some default setups and see if I get anywhere. I generally don't
have issues slicing any other model and wondered if there was something
specific about the geometry that was tripping something up, but you're
right, if it's working for you then it could potentially be something
specific to my setup.

It's a shame there's no verbosity or logging on failure, it briefly hangs
at 15% "generating walls" and  crashes the whole app.

Arakon, 2024-10-09:
Slices like a charm using your project file in 2.2.0 RC1, on Windows 11.

The-xyph, 2024-10-16:
A quick update... I removed all my presets and configs, and re-installed Orca (2.11) and it still crashes!

I have frequent crashes with most files, even with a fresh install... 

My Log has this entry.... 

Exceptions:
Exception Code :c0000005 ACCESS_VIOLATION
Exception Flag :0x0 NumberParameters :2 
Param 0 :0x1 
Param 1 :0x1888 
Context :0000009E92BF8C40 
ContextFlag : 0x10005f, EFlags: 0x10246 
Fault address:  0xFAA86805 0x1:0xA55805 C:\Program Files\OrcaSlicer\OrcaSlicer.dll

Show CallStack:

Fort completeness my specs are as follows:

CPU Vendor: GenuineIntel
CPU Brand: 13th Gen Intel(R) Core(TM) i9-13900K
CPU Family: 0x6
CPU Model: 0xb7
CPU Stepping: 0x1
CPU Type: 0x0
Speed: 2995 MHz
32 logical processors
24 physical processors
Hyper-threading: Supported
FCMOV: Supported
SSE2: Supported
SSE3: Supported
SSSE3: Supported
SSE4a: Unsupported
SSE41: Supported
SSE42: Supported
AES: Supported
AVX: Supported
AVX2: Supported
AVX512F: Unsupported
AVX512PF: Unsupported
AVX512ER: Unsupported
AVX512CD: Unsupported
AVX512VNNI: Unsupported
SHA: Supported
CMPXCHG16B: Supported
LAHF/SAHF: Supported
PrefetchW: Unsupported
Operating System Version:
Windows 11 (64 bit)
NTFS: Supported
Crypto Provider Codes: Supported 311 0x0 0x0 0x0
Video Card:
Driver: NVIDIA GeForce RTX 4090
DirectX Driver Name: nvldumd.dll
Driver Version: 32.0.15.6094
DirectX Driver Version: 32.0.15.6094
Driver Date: 8 14 2024
OpenGL Version: 4.6
Desktop Color Depth: 32 bits per pixel
Monitor Refresh Rate: 60 Hz
DirectX Card: NVIDIA GeForce RTX 4090
VendorID: 0x10de
DeviceID: 0x2684
Revision: 0xa1
Number of Monitors: 2
Number of Logical Video Cards: 2
No SLI or Crossfire Detected
Primary Display Resolution: 2560 x 1440
Desktop Resolution: 3640 x 1920
Primary Display Size: 23.50" x 13.23" (26.97" diag), 59.7cm x 33.6cm (68.5cm diag)
Primary Bus: PCI Express 16x
Primary VRAM: 24563 MB
Supported MSAA Modes: 2x 4x 8x
Sound card:
Audio device: Headset Earphone (2- Arctis 7 C
Memory:
RAM: 65276 Mb

The-xyph, 2024-11-06:
I've actually found I can get a more reliable slice (75%) after changing my system to use a manual virtual memory value to 20-32Gb...

closed 2024-11-06 by The-xyph