OrcaSlicer/OrcaSlicer #7015: Cannot edit text if generated in BBS first

DRA6N, 2024-10-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

If you create a project that has text within Bambu Slicer, and then open it in Orca, you can no longer edit the text. The Text becomes an object just like any other part.

In BBS:
[image]

In Orca
[image]

### Actual results

Cannot edit text layer in objects tab. 

### Expected results

Should be able to right click the text layer, and choose "Edit Text" But editable text isn't carried over from BBS to Orca. Only current workaround is to edit the text you need to change in BS, then import into Orca to print. 

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

avogra, 2024-10-05:
Orca uses a different tool for text than BambuStudio. So in that regard, they are not compatible. I would suggest to create text in Orca slicer, because the text tool there is much better.

dMopp, 2024-10-06:
That might be correct, but the File we download from makerworld.com are mostly (if not 100%) created in BBS. So somehow way to convert (at least once) the text would be a great option.

DRA6N, 2024-10-13:
> That might be correct, but the File we download from makerworld.com are mostly (if not 100%) created in BBS. So somehow a way to convert (at least once) the text would be a great option. (I used it for Lace Charms as an example.. "My" Workarround was, to modify text in bss and then open the file for printing in Orca..)

I have to do the same thing, some day I'll re-work it in Orca but for this specific project that I need to change the text quite a bit, I need the font to be the same, in the exact same location etc. Not an easy task to convert it all over to Orca, especially with multiple lines of text.

closed 2025-01-20 by github-actions[bot]